sharpie7/circuitjs1 #2: Short Circuit?

kenfink, 2015-07-20:
Two parallel wires seems to make it think there's a short circuit, despite a load in series with it.

http://lushprojects.com/circuitjs/circuitjs.html?cct=$+1+0.000005+10.200277308269968+50+5+50%0Av+176+400+176+80+0+0+40+5+0+0+0.5%0Aw+176+80+272+80+1%0Aw+272+80+368+80+0%0Aw+368+80+464+80+0%0As+272+80+272+160+0+0+false%0As+368+80+368+160+0+1+false%0As+464+80+464+160+0+0+false%0Ar+272+160+272+224+0+100%0Ar+368+160+368+224+0+400%0Ar+464+160+464+224+0+800%0Aw+272+224+368+224+0%0Aw+368+224+464+224+0%0Aw+368+256+368+224+0%0Aw+368+256+464+256+0%0Aw+368+256+272+256+0%0As+368+256+368+400+0+0+false%0As+272+256+272+336+0+1+false%0Ar+272+336+272+400+0+600%0Ar+400+448+400+512+0+200%0As+464+256+464+400+0+0+false%0Aw+176+400+272+400+0%0Aw+272+400+368+400+0%0Aw+368+400+464+400+0%0A

sharpie7, 2015-07-21:
Thanks for the comment. I shared it with Paul Falstad who wrote the simulation code. Basically the maths it uses won't work for this case (how will the current split between two zero resistance conductors?). Paul says "if there are any wire loops with no resistance (not just through the voltage source, but anywhere), the circuit matrix will be singular and unsolvable.  The trip around the two switches at lower right clearly qualifies.".

jonpatterns, 2016-03-22:
A possible solution to this would be to assign a very low default resistance value to wires and swtiches, that would be consistent with reality too. However, I'm not sure how the maths function would handle this - for example it may require good decimal accuracy.

hausen, 2016-04-07:
If it helps, in my branch of Paul Falstad's original simulator, I have made WireElm a subclass of ResistorElm to work around this isse. See: https://github.com/hausen/circuit-simulator/blob/master/src/WireElm.java

It causes some side effects, though, so I had to add an "ideal wires" option to the menu. The user may select which behavior she wants.

jonpatterns, 2016-04-07:
Interesting, I'll take a look when I get a chance.

Harvie, 2016-08-14:
Nice! I always thought that wires should have at least very tiny resistance to stabilize the simulation...

Harvie, 2016-08-15:
Also lot of times there are no "wires" in circuit, there are just multiple components bonded directly by it's pins. so "global wire resistance" should be applied to all component pins in the first place. once this will be done, the "wires" will not really need it as they will always end in some component with series resistance of it's own.

sharpie7, 2016-10-16:
The convention that wires in circuit simulators are ideal is pretty well established and benefits in consistency with components directly joined and helps the performance. Real wires are not simple resistors but also have inductive and capacitative effects (and perhaps even non-linear in extreme cases). If non-ideal performance of wires is critical to the design then users should create their own model.

Therefore wires as ideal will be retained as the behaviour for now.

closed 2016-10-16 by sharpie7

Harvie, 2016-10-16:
But there are cases where simulator fails to converge or throws error because of this.