sharpie7/circuitjs1 #3: Resize circuit

mannanj, 2015-08-28:
Hi is it possible to resize the simulator, I can't seem to do it by adjusting the width/height in the iframe. Would this be done Java side?

Sincerely,
Mannan

sharpie7, 2015-08-29:
If the simulation is contained in an iframe it should adjust to the size of the iframe unless the iframe is too small. For example:
<iframe src="circuitjs.html" width="816" height="640">

From http://lushprojects.com/circuitjs/

The page circuitjs.html will always fill any iframe that contains it, but the iframe doesn't have to be full screen.

mannanj, 2015-08-31:
Beautiful. I got it to work by having an iframe link to the circuitjs.html
```
<iframe src="/home/samsung/Documents/Projects_Active/omf/omf/static/war/circuitjs.html" height="640" width="816"> 
```
Thanks so much!

sharpie7, 2015-09-04:
Great. I think that closes this issue!

closed 2015-09-04 by sharpie7