sharpie7/circuitjs1 #9: Unrestricted amps drawn with LEDs and Inductors in circuit with less than one ohm

jonpatterns, 2016-03-22:
As shown in the demonstration video, and LED in a DC circuit with 0 ohm resistance will draw an extremely large current.
 
https://youtu.be/o4iGcNkldEY?t=10m42s

An inductor in a DC circuit with 0 ohm resistance, the current drawn will keep increasing - beyond what could realistically be supplied. The same is true if the resistance is less than one ohm.

Perhaps a simple solution is to add a max current attribute to power sources - it could even graphically signal its at maximum current.

Below is a test example that can switch 0/0.5/1 ohm and LED / inductor.

```
$ 1 0.0000049999999999999996 382.76258214399064 34 5 50
v 176 448 176 320 0 0 40 5 0 0 0.5
w 720 384 720 544 2
w 720 224 720 384 1
l 528 560 624 560 0 1 -8.481532894164689
r 272 240 368 240 0 0.5
S 176 224 272 224 0 1 false 0
w 400 240 432 224 0
w 400 208 432 224 0
w 368 208 272 208 0
w 368 208 400 208 0
w 368 240 400 240 0
w 624 240 656 240 0
w 624 208 656 208 0
w 624 208 528 208 0
w 656 208 688 224 0
w 656 240 688 224 0
S 432 224 528 224 0 0 false 0
r 528 240 624 240 0 0.5
w 688 224 720 224 0
162 368 560 272 560 1 2.1024259 1 0 0
w 240 560 272 560 0
w 240 528 272 528 0
w 368 528 272 528 0
w 208 544 240 560 0
w 208 544 240 528 0
S 464 544 368 544 0 1 false 0
S 720 544 624 544 0 0 false 0
w 464 544 496 528 0
w 464 544 496 560 0
w 624 528 528 528 0
w 496 528 528 528 0
w 496 560 528 560 0
w 176 256 176 224 2
w 176 544 176 512 2
w 176 544 208 544 0
w 176 320 176 256 1
w 176 448 176 512 1
```

CapacitorSet, 2016-03-22:
The simulator deals with ideal voltage sources, and they are expected to
supply an infinite current. If you want to simulate a real voltage source
there are several models, starting with Thevenin.

jonpatterns, 2016-03-22:
Thanks for the reply. So putting a resistor in series with voltage source will approx. real voltage source.

sharpie7, 2016-10-16:
Closed - as explained by @CapacitorSet the simulated behaviours are for ideal voltage sources.

closed 2016-10-16 by sharpie7