sharpie7/circuitjs1 #10: Switch double click

Saurbaum, 2016-04-07:
I can see why making a double click on components as a way to edit them is a good idea.  The problem is you might click on switches more frequently.

It's driving me crazy when I'm tweaking things that if I click on a switch to fast it popups up the edit dialogue.  It's even more likely to happen with a momentary button.

billcollis, 2016-04-07:
I also found this a little bit of a nuisance at times so I changed the method in CirSim.java that handles double clicks
onDoubleClick(DoubleClickEvent e)
add 2 lines
if (mouseElm instanceof SwitchElm) 
   return; 
It means that users now have to right click a switch to change ts options

sharpie7, 2016-10-16:
Fixed in "one-year-updates" branch. Will merge to master in due course.

closed 2016-10-16 by sharpie7