sharpie7/circuitjs1 #13: How to check Circuit state after create a circuit?

QFord, 2016-06-23:
When create a circuit and failed ,it's necessary to know circuit wrong state. Especially tell user why fail to create right circuit.
Is the a method to do this ? Thx.

billcollis, 2016-06-23:
I am working on this, it will be a little while till its ready
Bill

QFord, 2016-06-23:
Thx,this is really a good open source circuit simulation.
BTW, I found a bug when I exec ParallelVoltageSource.
I used a C# version which translate by a kind guy.

`private static void ParallelVoltageSource()
        {
            Circuit sim = new Circuit();

            var volt0 = sim.Create<DCVoltageSource>();
            var volt1 = sim.Create<DCVoltageSource>();
            var res0 = sim.Create<Resistor>();

            sim.Connect(volt0.leadPos, res0.leadOut);
            sim.Connect(res0.leadIn, volt0.leadNeg);
            sim.Connect(volt1.leadPos, res0.leadOut);
            sim.Connect(res0.leadIn, volt1.leadNeg);

            sim.doTick();
            string s1 = string.Format("V{0} = I*R {1}*{2}", res0.getVoltageDelta(), res0.getCurrent(), res0.resistance);
            Debug.Log(s1); // V = I x R

            Console.WriteLine("program complete");

        }`
Now, I developed an Unity3D electric simulation demo for my students.
I made a VR version, it seems very interesting, I named it VRLab.
I've google for some days,but frustrating, there r few electric simulation engine that can be used directly by Unity3D.

limitedAtonement, 2020-10-21:
I think this can be closed. There is currently a "bad connection" notification on the lower right. This will likely meet the original poster's needs.

pfalstad, 2020-10-21:
I don't quite understand the issue but I'll close it.  It seems to be a bug in someone's C# version anyway.  This is not the place to report bugs in that.

closed 2020-10-21 by pfalstad