sharpie7/circuitjs1 #14: Add TRIAC, Rail to rail OPAMP, TL431, DPDT switch and LM317, LM337, 7805

Harvie, 2016-08-14:
Hello,
can you please add following components?

TRIAC (it does not work very well when i simulate it using two SCRs). This can be very usefull.

Rail to rail OPAMP with power supply pins (currently the OPAMP in circuit does not allow to connect supply voltage). So it's output is not dependent on supply voltage of the rest of the circuit, which caused me problems in past.

TL431 quite common component that integrates precision voltage reference, opamp, NPN transistor and diode. They are almost everywhere nowadays...

DPDT switch. Also quite usefull. Now i have to use single switch and relays.

LM317, LM337, 7805 - quite favourite linear regulators are something that i miss in simulator. They are also very popular. Maybe it can be just single 3-legged component with configurable polarity, voltage reference and voltage drop. defaulting to LM317.

pfalstad, 2016-08-14:
DPDT switches are just two DPST switches that are linked.  Create two DPST switches, select File->Export As Text, copy the text and paste it into an "Import As Text" window.  find the two lines starting with "S", and change the last number to a "1" (or any other unique number if you need multiple DPDT switches).  I don't know why I haven't made a simpler way to do this.

Harvie, 2016-08-14:
Maybe just add "switch id" or something like that to right click edit menu... However this is not my biggest pain. I struggle much more with triacs and opamps...

eigengrau, 2017-01-10:
Maybe some sort of varicap might also be something for the wishlist.

sgtwilko, 2017-07-09:
I too would be interested in the PC817 Opto-isolator.  
I have had a look, but can't see documentation on adding new components types.

servicemeson, 2017-11-13:
Sorry,  I'm not an expert.
I used a model of the optocoupler 4N25 , Current Transfer Ratio (CTR) =20%, 
 transfer output function is -0.2*i
optocoupler&pnp_darlington.circuitjs.txt

Import from text.

$ 1 0.0000049999999999999996 222.68277840103312 99 5 43
w 224 96 320 96 0
s 320 96 448 96 0 0 false
w 448 96 496 96 0
w 496 96 496 128 0
v 496 272 496 128 0 0 40 24 0 0 0.5
w 496 272 496 368 0
O 288 288 288 192 1
w 160 240 160 224 0
w 160 144 160 96 0
w 160 96 224 96 0
w 160 240 128 240 0
w 128 240 96 240 0
w -64 368 -64 336 0
w -64 336 -32 336 0
w 64 336 96 336 0
w 96 336 96 400 0
w 96 400 224 400 0
w 224 400 496 400 0
w 496 400 496 368 0
w 64 304 96 304 0
r 96 304 96 240 0 1000
400 160 240 224 240 0 0_-1_-0.5886099676151062_-0.5915206143466136_100 0_-1_0.002910646731507427_-0.6550118902456354_100 -1
w 224 224 224 96 0
r -96 304 -96 384 0 10000
w -64 368 -64 400 0
w -96 384 -96 400 0
w -96 400 -64 400 0
g -64 400 -64 432 0
w -96 304 -96 256 0
w -96 256 -96 240 0
g 224 400 224 432 0
v -160 400 -160 240 0 2 40 2.5 2.5 0 0.5
w -160 400 -96 400 0
s -160 240 -96 240 0 0 false
215 -32 304 16 304 0 2 -0.2*i
r -96 240 -32 240 0 390
d -32 240 -32 304 1 1.3
r 160 144 160 224 0 1000
w 224 256 224 272 0
w 224 272 224 288 0
r 224 400 224 336 0 2400
d 224 288 224 336 1 1.3
r 288 288 224 288 0 30
o 6 64 0 4098 40 0.1 0 1

sharpie7, 2019-05-01:
Triac and optocoupler has been added. Normal policy is not to add specific device models unless they are really iconic elements (555, 741).

closed 2019-05-01 by sharpie7

Harvie, 2019-05-01:
> Triac and optocoupler has been added.

That's great news! Thank you.
Just tested the optocoupler and it looks nice!

> unless they are really iconic elements

While i think that TL431 and LM317 or 7805 are almost iconic components found in wide variety of circuits. I dont really need exact component model. It might as well be generic component with configurable parameters. However i guess i can build TL431 using individual components and i usualy replace voltage stabilizer using 2 pin voltage source. Which is not always ideal, but somewhat works in simple cases.

DPDT switch and possibly full bridge rectifier would be nice to have for convenience.

But not having supply voltage pins at opamp is very hard to overcome in circuits, where the opamp supply voltage fluctuates for some reason (simple example is low voltage disconnect circuit, where opamp is used to check it's own supply voltage, i've simulated it in simulator, but real world circuit was behaving very differently due to opamp supply voltage not being constant).

pfalstad, 2019-05-02:
DPDT switch is there, just create two SPDT switches, edit them and set the switch group of each to the same nonzero number.

Harvie, 2019-05-02:
Didn't realized there's right click UI for switch groups now... Nice!

What about the opamp problem? Do you think it would be possible?

pfalstad, 2019-05-02:
Sure it’s possible..  we have the entire 741 as one of the sample circuits, in fact.  Would be nice to simplify it for speed though.  This is high on my list.  Also it’s filed as a separate request, #330 .

pfalstad, 2019-05-02:
On second thought why not just use the 741?  That way we'll be sure it's accurate.  If people want something faster they can use the ideal op-amp.

Harvie, 2019-05-02:
> On second thought why not just use the 741? That way we'll be sure it's accurate.

Because of the cognitive load...  look at the circuit:

[image]

Imagine that i will be explaining some complex circuit containing several opamps to students (or hell! even to myself) using simulator. That's like forrest of transistors stealing the attention from the circuit itself. Also editing such circuit might became a nightmare if things get too tangled.

BTW This is exactly why i proposed that it would be nice if it was possible to nest the circuits. So that we would be able to take the 741 example, define some interfacing pins into it and then import it as "black box" module into another circuit without having to cope with it's complexity.

There can even be public library of user contributed components for other people to reuse.

Surely this will not be super fast and optimized, but maybe we can even manage to make some "compiler" which will automaticaly create fast flat numeric model from such contributed module.

pfalstad, 2019-05-02:
I meant to say, why don’t I implement the improved op amp component by encapsulating the 741, as you say, rather than trying to simplify it somehow.

Harvie, 2019-05-02:
@pfalstad Oh i see :-) What do you think about possibility to add user UI to create custom reusable "sub-circuits"?

pfalstad, 2019-05-03:
Yes, that would be good.  It might not be so hard.  Iain has already done a big chunk of the work.