sharpie7/circuitjs1 #15: Copypaste between two different windows, paste import, compact circuit blocks?

Harvie, 2016-08-14:
Hello,
i think that copypaste between two different windows would be nice feature :-)

Also it would be really usefull if importing schematic from file would allow me to import one schematic to another (like copypaste, but from file), so i can import reusable circuit blocks to bigger composite circuit.

Another step might be adding support for real blocks: Eg. i make full bridge rectifier, add "pin components" with labels to it's endpoints and save it to file. Then i "import as block" which would show whole rectifier as single component with 4 pins. With connection being dynamic, so i can modify the block without breaking the super circuits that are using such blocks.

pfalstad, 2016-08-17:
I added copy paste between windows to my branch

sharpie7, 2016-10-16:
Fixed in "one-year-updates" branch. Will merge to master in due course.

closed 2016-10-16 by sharpie7

Harvie, 2017-03-28:
What about the blocks?