sharpie7/circuitjs1 #18: LED brightness vs current curve

endolith, 2016-08-31:
In reality, I have an LED with 0.3 mA going through it, and another with 1.4 mA, and they're plenty bright.  I can't simulate this, though, because in the simulator, it looks like you need 10 mA to get full brightness, and it quickly dims to invisibility below this:

[image]

Is there some way to make the LED more realistic?  I'm not sure if this is because the brightness vs current curve should be nonlinear or if it's just efficiency differences between different LEDs, but I'd like to be able to have the LED at full brightness when the real LED is at "full" brightness, rather than using inaccurately small resistors in the simulation that then don't behave like reality.

pfalstad, 2016-09-08:
Would it be enough to have a LED parameter to specify the current at max brightness?  This will vary depending on the LED.

pfalstad, 2016-10-04:
fixed on my branch

sharpie7, 2016-10-16:
Fixed in "one-year-updates" branch. Will merge to master in due course.

closed 2016-10-16 by sharpie7

pfalstad, 2016-10-29:
Ok I've been playing with some real LED's and it seems that the brightness should be a logarithmic function of current, not just linear.  I'll have a fix soon.