sharpie7/circuitjs1 #21: EKV mosfet model

hatsunearu, 2016-09-19:
So I'm taking this class called ECE 3400 at Georgia Tech with professor Jennifer Hasler, and her class is about making analog systems with subthreshold mosfets. That's right, the leakage current isn't just a nuisance, but an important low power analog circuit building tool. I want to create some subthreshold-compatible mosfet elements for this project, and I just want to announce it here. Hopefully I can get it integrated in the original website, because your sim really helped me learn the basics.

pfalstad, 2016-11-09:
This sounds interesting.  But do you need to change the mosfet models?  Or can you just build circuits that have standard mosfets operating below the threshold voltage?

Let us know what happens with this.

hatsunearu, 2016-11-09:
probably gonna make a new circuit element. I've read through the code but things are pretty convoluted. do you have any documentation on what some of the variables/classes mean?

pfalstad, 2016-11-09:
Not really.  The code you want is in MosfetElm.java.  I can help you add a new element or make modifications to the existing one.  But I still don't see why you need one.  What is the real-world device you are simulating?  And how does it differ from a regular MOSFET?

hatsunearu, 2016-11-09:
well I was thinking of duplicating the original model and using the ekv model to get that working. the cool thing about subthreshold is the lower power usage and much better transconductance and compliance when used as a current source. 

It turns out i won't have time to work on this for a while, so it might be better to table this for now.

pfalstad, 2016-11-09:
OK!  I get it now.  Sorry.

Yeah, the MOSFET model in this simulator is very simplified and is very unrealistic in the subthreshold region.  It would be interesting to see an EKV model based MOSFET element.  Might be best to have a flag on the MOSFET element to specify which model to use, rather than a whole new element.  Or a subclass at least.  Anyway, if you get time to work on this, let me know if I can help.

hatsunearu, 2016-11-09:
One more thing: in the subthreshold region, it's better to have access to the bulk/body of the MOSFET because that gets you higher transconductance and better analog behavior, so I'm not sure if using the same element is a good idea...

pfalstad, 2016-11-09:
Yeah, in that case maybe a new element.

E3V3A, 2019-02-26:
@hatsunearu 
[...almost 4 years later...]
This sound really cool. Can you provide us some references to this work?

hatsunearu, 2019-03-01:
..Whoops!

I tried looking at the source code (4 years ago...) but it was way too daunting (haven't touched Java in like 8 years) so I gave up.

Here's some resources if anyone wants to have a go at it. 

http://hasler.ece.gatech.edu/SoCFPAA/L2Simulation_JAICSP.pdf

closed 2019-03-01 by hatsunearu

E3V3A, 2019-03-01:
> haven't touched Java in like 8 years
That's how they determine the Java version numbers...how many years it will take until you wanna use it next time.