sharpie7/circuitjs1 #26: A Few Suggestions (Implementations/Components)

bhottovy, 2016-11-02:
First off, I just want to say I am glad this project was picked up again. The changes already make it work much smoother (Component organization, especially the crosshairs so it isn't difficult to line up wires, being able to drag components from switch nodes), and I can't wait to see how much better it gets. 

Anyways, onto some of my suggestions. I definitely like some of the ideas listed by Harvie (Triacs, Isolators, full bridge rectifier, etc), and thought I might as well suggest a few of my own. Don't know if anyone else thinks they are particularly useful or how difficult they might be to implement, but here they are:

**Fuses**: I'm actually surprised this was never added. Just a simple component that breaks when a certain current is reached, for testing of protection circuits.

**Varistor**: Isn't exactly reproducible by anything else, and would add a lot more functionality to circuits that wasn't possible.

**Cable**: Don't know if anyone else would find this remotely useful, but I thought it could be in some instances. While the wire is a short, a cable could simply be a wire over a significant distance. While a resistor could easily be used in this instance, I just thought this could be a simple way to test wiring, by simply changing values of conductivity, radius and diameter to get different resistances (or even just have a drop-down list of different gauges and wire materials).

**Multi-point switches**: Not sure how difficult this would be to implement let alone deciding node placement, especially since the program is based off a square grid. This would be rotary switches, or switches that have a single input, with a changeable amount of outputs (say, between 3 and possible 10).

**Interactive probe**: Works like the place-able scope probe. The idea is you would place down a lead for a probe, and then the cursor would display a voltage difference between the placed probe and the component/node the cursor is hovering over. Could be enabled like a mode, where simply clicking again would replace the reference point.

**Oscilloscope Tweaks**: Includes showing time intervals of the tick marks, allowing for vertical tick-marks that show the voltage/current/power intervals for each oscilloscope separately (currently they are automatic, and that can get annoying sometimes when the graph is close to zero and indistinguishable, or when a sudden spike can blow the graph up)

**External Waveform Input**: This would probably be extremely difficult and/or time consuming to implement, but I will go ahead and suggest it. This would allow a custom waveform to be imported somehow, either from a certain file type or a built-in waveform editor. This could allow testing of more complex filtering circuits.

~~**Logic Table**: Could work similar to the built-in oscilloscope. With a series of logic inputs and logic outputs, a user could be shown a logic table (assuming there is no time dependence, other than a clock, which could simply all be connected as 1 logic input). Not sure how it could be implemented, but thinking the user could select all the desired logic inputs/outputs, the program would run through every possible combination and display a table.~~ -Right after writing this, I figured this might be pointless, as there is plenty of software and programs already out there for pure logic circuits that it wouldn't be worth adding. However, I'll leave it here in case it might still be implemented.

**Motors**: (May be impossible to estimate, and completely dependent on the exact part) Would allow the user to input an RPM and stall torque, and see the effect a given load would have on the circuit (And see an estimated relationship between the output torque/rpm). Can help for designing motor/actuator interfacing and power management, especially H-bridges and such.

**Internal Resistance**: Just a simple tweak that would add an optional internal resistance to a voltage source. Not at all necessary, but convenient.

**Solar/Light Dependent Circuits**: A setting to set the ambient light (in lumens, lux, or whatever else would work best) could be added to options. This would allow for the application of photoresistors/transistors, solar cells, and testing them in circuits. The light created by components could be ignored.

**Symbol Tweaks**: First off, I think an option to edit grounds and toggle between a chassis and earth ground symbol (for each one seperately) could be added for easier identification. Also, like the european resistor toggle, an option could be added to show the bulk arrow on the mosfets.

Hopefully this list proves helpful in some way. I already had some second-thoughts about some of the ideas I posted, but I'll leave them in anyways just in case.

pfalstad, 2016-11-03:
Good suggestions, thanks!  I don't quite understand interactive probe or the scope tweaks.  The scope has an option to display the scale.  I have some code ready to show the time and voltage/current values at the mouse pointer on the scope.  I also have some code written to change the default mosfet symbol to be the more common one with the bulk terminal.

adrians, 2018-01-03:
Hello,

In a development branch I changed the Ground symbol ([link](https://github.com/adrians/circuitjs1/commit/5cd9e572210e53548870609f10bfd9262f37af16)), such that instead of the Earth-GND, it uses the Circuit-GND symbol.

Current Earth-GND symbol:
[image]

My Circuit-GND symbol:
[image]

Basically, there are 3 main ways implement this change:

1. Replace the old symbol entirely.
2. Add a toggle, like for the EU/US resistors.
3. Implement a new component that does the same thing as the previous GND symbol, but looks different. New circuits will be able to contain both Earth-GND and Circuit-GND symbols (doing the same thing).

@sharpie7 : Which way would you prefer, such that it gets merged into the main repo?

pfalstad, 2018-01-04:
I think you should be able to right-click and edit the ground component to change the symbol. There's at least 4 different ground symbols that could be used, so having a new component for each one would be bad.

limitedAtonement, 2020-10-21:
This can probably be closed as it's been inactive for nearly three years. @adrians If you are able, please submit the PR with an option to use use either symbol as pfalstad mentioned?

@bhottovy If you want to suggest more of these features, be sure to check the https://github.com/sharpie7/circuitjs1/blob/master/CONTRIBUTING.md document and open one ticket per feature request so they can be worked on properly. Thank you!

pfalstad, 2021-12-11:
ok, I think everything is done or filed separately in this list except for cable and interactive probe.  Varistor is #332.  External input is #309.  Internal resistance is #677.