sharpie7/circuitjs1 #27: Last update broke things.

Erquint, 2016-11-03:
Just want to report overall stability and accuracy deterioration with the latest update(-s).  
I'm getting GA and GV of dormant current for one and not even stopping and/or resetting fixes it.  
Could you make a separate page for the old stable version?

pfalstad, 2016-11-03:
Hi, could you be more specific?  Could you send us the circuit, using File->Export as Link or File->Export as Text, and explain what's wrong with it?  Are you talking about the version at falstad.com or lushprojects.com ?

Erquint, 2016-11-03:
I didn't know there are versions other than at falstad.com.  
Well, I think the main problem is convergence.  
I had it running with "convergence failed" message on screen at one point.  
Well, let me try mashing something up and I'll send you the link if I can before going out(6:30AM here).

Erquint, 2016-11-03:
Hmm, and now GitHub's formatting is failing on me.
That message above isn't supposed to look like I have no idea what Markdown is.
Doesn't look like that in the preview at least.

Here's a [stupid little circuit](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+0.008208499862389881+99+5+43%0Av+656+336+656+416+0+0+40+5+0+0+0.5%0Av+736+416+736+352+0+0+40+5+0+0+0.5%0At+656+336+688+336+0+-1+-10.000000000000004+-9.982671320486002+100%0Aw+736+352+688+352+0%0Aw+688+320+768+320+0%0Aw+768+416+736+416+0%0Aw+736+416+656+416+0%0Ag+736+416+736+432+0%0A162+768+320+768+416+1+2.1024259+1+0+0+0.01%0A). Take notice that the simulation speed is one step up from 0. Look at the ridiculous numbers. Now if you put simulation speed to 0 and try disconnecting the circuit and resetting the simulation — conductors remain powered.  

As an unrelated note: can you please make transistor notation setting saved to cookies?

pfalstad, 2016-11-03:
Hi, I tried it on the original java version and got a similar result.  There were ridiculous numbers, which seems right.  So I moved the transistor out of the circuit and it still had huge current numbers.  Definitely a bug, but the java simulator had the same problem.  I have not tried it on an old version of the javascript version.

I tried resetting after disconnecting the transistor and everything seemed fine.

Yes, I plan to make resistor notation saved by cookies.  Right now it looks at your browser language setting.

Erquint, 2016-11-03:
[Caught it](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+0.9487735836358526+1+5+43%0Av+656+336+656+416+0+0+40+5+0+0+0.5%0Av+736+416+736+352+0+0+40+5+0+0+0.5%0At+656+336+688+336+0+-1+-5.247663124966176+-5.230334445452147+100%0Aw+736+352+688+352+0%0Aw+688+320+768+320+0%0Aw+768+416+736+416+0%0Aw+736+416+656+416+0%0Ag+688+320+688+304+0%0A162+768+320+768+416+1+2.1024259+1+0+0+0.01%0A). This little modification made the circuit run with "convergence failed" message on screen if you press "Run/Stop" button and the current flow is pretty wanky.

Erquint, 2016-11-03:
[The unresettable bug](https://gfycat.com/ParchedHairyAmericanrobin) only occurs with 0 simulation speed.

pfalstad, 2016-11-03:
Ok I tried the "caught it" circuit in the java version and also got a "convergence failed" there.

If the simulation speed is set to 0, the simulation stops.  Lots of weird things happen if you move the circuit elements around.  You can even try this with the LRC circuit.  The simulation never gets run to update currents.  If you press reset, it resets the voltages but not currents.

Anyway, I definitely see some issues that need fixing here.  But I don't think any of them are new problems.

If you try to put a voltage source across a bare transistor like that, it should probably just give an error rather than report an unrealistically large current.

sharpie7, 2017-01-16:
I believe Paul's revisions have fixed this in the JS version now, so closing this issue. thanks for reporting.

closed 2017-01-16 by sharpie7