sharpie7/circuitjs1 #28: Wish: tinyurl

cehteh, 2016-11-03:
add export as shortened  (maybe with a local database hash:full_link instead external service)
This would make collaborative editing/pasting easier.

Chisight, 2016-11-03:
Alternatively, a load from URL option in the URL might be easy to implement and could leverage existing pastebins.

sharpie7, 2016-11-03:
I've never been keen on "load from URL" because it's a soft-off cross-site scripting situation. What I would like to do one day is to integrate with something like Google Docs for cloud-based file support.

Chisight, 2016-11-03:
If not load from url, maybe pick a pastebin to support.  GoogleDocs comes with a whole bunch of tracking and other issues, a raw pastebin should be much less code to write.

pfalstad, 2016-11-03:
I looked into this but ran into a browser security issue.  Our javascript can't access other websites because of the same-origin policy.   So you can't use pastebin's api.

One way around that might be to provide an option to store the circuit data in JSONP format.  But you'd still have to paste it manually.

I might be willing to provide a server to store the circuits on my website.

sharpie7, 2016-11-03:
When I looked at this before I concluded that some kind of server-side feature was needed either to support local storage or to escape from the origin policy of the browser to access third-party services.

Using a database might be technically workable though I am not sure I like the statistical purging approach. Generally, looking after other people's data is not something I am keen to do for several reasons.

Christian, why don't you like the approach of using existing cloud services?

Iain

sharpie7, 2016-11-03:
BTW - I agree that a partial solution might be to have a compressed circuit format for URLs supported. GZIP might work but there is a lot of structure in the files that, if exploited, could probably achieve a much better result than a general compression algorithm.

cehteh, 2016-11-03:
> When I looked at this before I concluded that some kind of
> server-side feature was needed either to support local storage or to
> escape from the origin policy of the browser to access third-party [...]

What are the reasons? I understand that such a thing comes with some
responsibility. But for something like this schematics I don't see too
much potential this could be abused, neither traffic nor content wise.
Of course you should be clear somehow in the terms of service that you
are not responsible for any uploads and may remove them for whatever
reason. While ideally it should be a no brainer and just work. Thats
why I propose this automatic scrubbing. Even if you set a default TTL
to some years, any unused (spamed) content will eventually be
purged. Publishing user content in the menu or make it
search/indexable needs to be taken with a grain of salt, I am aware of
that. But having shortended links only on demand/invite should be
mostly harmless.

btw: sqlite was just an easy idea. This is actually so simple that it
could be stored in the filesystem and served statically. One dir per
hash, schematic and metadata as file, title as symlink and scrubbing as
simple shell script using unix times.

> 
> Christian, why don't you like the approach of using existing cloud
> services?

like you saied, b/c all that tracking BS almost all do, often one
is required to have an account there. And possibly you force anyone who
you share the things with into that too. Also you get multiple points of
failure, dependencies and any 'free' cloud service may decide to stop
its operation at any time for any reason.
This might be true for yours and Pauls website too. Who knows about the
future. But in spirit being an open source project I could at least
establish my own website then. I am not doing so yet because there is
no need, I like to use your online versions and I think you deserve the
fame for this project. Dunno how much traffic these sites cause and if
they cost you anything more or even generate some profit for you. For
me its just useful to have these services, thank you.

Either way, having some pastebin/tinyurl like way to share circuits
would be good. Exporting them as URL is way too long for example to
paste in IRC or in an email. So far i used tinyurl.com for that but its
a bit of a hazzle to copy paste things around and eventually such links
get lost unless i bookmark them for myself.

	Christian

sharpie7, 2016-11-06:
I have added integration with a URL shortener as part of the "Export as link" dialog. Live now on my version: http://lushprojects.com/circuitjs/

Dunno if I really want to add being a cloud storage provider, even in a very limited way, to my many activities. Supporting *anything* is always more work then you expect in my experience. I have added integration with Dropbox because their API is nice and high-level and compatible with a client-based solution.

When I get time (maybe later today) I plan to add support for Dropbox links in the query-string parameters and also add some file naming support to the circuits.

sharpie7, 2016-11-06:
?startCircuitLink paramter added to the query string. The load is done from the client without touching the sever so the link server must be CORS compatible. Dropbox links are supported.

cehteh, 2016-11-07:
> I have added integration with a URL shortener as part of the "Export
> as link" dialog. Live now on my version:
> http://lushprojects.com/circuitjs/

awesome! thank you very much!

	Christian

> Dunno if I really want to add being a cloud storage provider, even in
> a very limited way, to my many activities. Supporting *anything* is
> always more work then you expect in my experience. I have [...]

i don't use dropbox, but might be useable for some folks :)

sharpie7, 2017-02-26:
Closed as equivalent functionality now included.

closed 2017-02-26 by sharpie7