sharpie7/circuitjs1 #30: Connector support?

JamesCoyle, 2016-11-07:
I'd personally find it useful to have support for connectors. These would act like multiple switches that all toggle on and off at the same time. It could be a simple thick-ish line that crosses over the wires that when clicked on splits into two thinner parallel lines to show it is disconnected.

[image]

As a bonus it would be awesome if it was possible to simulate the connections being shorted together (as is the case with 3.5mm audio connectors) or wires disconnecting at slightly different times (as can happen with many of the long multi-pin connectors).

sharpie7, 2016-11-08:
It's not obvious unless you know it, but SPDT switches can be assigned groups which gang them together so that they switch simultaneously. This could be used to simulate the electrical behaviours that you want, though not as pretty as your solution.

limitedAtonement, 2020-10-21:
This can probably be closed as it's almost four years old. @JamesCoyle Please feel free to create a PR with your cool-looking multi-pole switch!

pfalstad, 2020-12-13:
I think SPDT switches are the way to go here.

closed 2020-12-13 by pfalstad