sharpie7/circuitjs1 #33: Element leads do not connect correctly

hondero2552, 2016-11-09:
I found this bug while using the app to build a circuit. After you move elements around a couple of times, the leads will stop connecting
[image]

. In fact, you cannot connect them at all. Seems something must be wrong with the coordinate system when dragging the leads of any element. 

I am not very familiar with your code but I will try to look into this myself.

I will also try to give step by step instruction to reproduce the bug, specifically using one of the built in circuits.

I have attached an image. The leads of the individual elements cannot be connected again unless you create a new elements. I'm thinking it has to do with going from screen size changes.

pfalstad, 2016-11-09:
Step by step instructions would be great.  If not, what sorts of things were you doing?  You were just moving elements?  How were you moving them, just dragging individual elements?  Dragging the whole element, or the terminals?  Do you have Options->Small Grid turned on?  Maybe turning that on will help.

hondero2552, 2016-11-09:
I was doing all of the above. Dragging elements around, moving them both by the terminals and as a whole element. I wasn't doing anything crazy really but it's the second time this happens to me.

I did not turn on or off the small grid settings. In fact, I didn't change them at all.

sharpie7, 2017-02-26:
Closed as unable to reproduce.

closed 2017-02-26 by sharpie7

hondero2552, 2017-03-05:
Alright, fair enough. I started working on another project so I won't be able to troubleshoot this any more.