sharpie7/circuitjs1 #39: Unwanted node "dots" added to circuits

sharpie7, 2016-11-21:
@pfalstad Somewhere in the recent changes (I guess due to the "wire" optimization) the sim has started drawing node dots in places where they shouldn't be (example attached). There is a node there, but it wouldn't be marked with a dot in a circuit diagram. I think this worked correctly before.

I haven't gone hunting for the fix yet, but I guess it might ring a bell with you and you could go straight to it.
[image]

pfalstad, 2016-11-21:
yeah, i see what's causing this..  hmm, will have to think about how to best fix it.

pfalstad, 2016-11-22:
fixed in my branch, thanks for noticing that.

jrao, 2016-11-23:
I'd like to build the application but I ran into problems setting up the project with the Eclipse GWT plugin. It seems that the required App Engine SDK is not getting installed. I might also have a Java version incompatibility problem. Could you update the README with the required Java version, Eclipse version, and Eclipse GWT plugin? Any other dependencies? (Sorry, not sure the best place to ask this.)

sharpie7, 2016-11-24:
Thanks Paul - great work!

closed 2016-11-24 by sharpie7