sharpie7/circuitjs1 #41: Javascript version has started to use much more resources and is generally unusable now

emilfihlman, 2016-12-02:
http://www.falstad.com/circuit/circuitjs.html used to work without any issues but in the last month (max time the effect started) the site is now way more resource intensive and generally unusable because of lag.

pfalstad, 2016-12-03:
Could you be more specific about what seems slower?  It seems like it should be faster.  Is there a particular circuit you're working on that is slower?  Is the simulation slower, or something else like adding components, or....?

pfalstad, 2016-12-03:
I guess you were mainly talking about resource use and not speed.  Are you working with a large circuit?  It still seems like use of resources should be less if anything.

emilfihlman, 2016-12-03:
Hi, yeah resource usage, which in terms makes it very laggy :/

Here's an example of a circuit that runs very slow, no matter where I run it:
$ 1 0.000005 0.027481238055948966 50 5 43
R 432 144 400 144 0 0 40 240 0 0 0.5
f 608 192 608 144 17 1.5 0.02
f 496 256 608 256 16 1.5 0.02
r 544 144 544 192 0 100000
w 544 144 592 144 0
r 704 144 704 336 0 100
w 608 192 608 240 0
w 608 272 608 336 0
z 496 336 496 256 1 0.805904783 3.3
w 608 336 496 336 0
w 704 336 672 336 0
w 672 336 608 336 0
w 704 144 672 144 0
w 672 144 624 144 0
g 496 336 416 336 0
r 496 144 496 256 0 10000000
s 432 256 496 256 0 1 false
w 608 192 544 192 0
g 432 256 416 256 0
w 496 144 544 144 0
r 432 144 496 144 0 100
o 5 8 0 2083 160 1.6 0 -1 0

I can shoot a video of the behaviour on a couple of computers but that has to wait until tomorrow.

I hope this gets resolved, I use the simulator quite a lot and it has never before acted like this.

pfalstad, 2016-12-05:
hi, I'm still not seeing it.  what platform are you using?  what browser?

modwizcode, 2016-12-07:
Hi, I'm seeing this too, specifically in chrome which seems to start consuming a major amount of resources even on loading the front page (i.e. http://www.falstad.com/circuit/) it definitely was not doing that as of a few days ago in my case. Whatever the site is doing it's overwhelming chrome but not edge. I often find the whole chrome process group freezing when garbage collection occurs.

I've attached some profiling from chrome, just from running on the front page with a medium-low load on the computer. Something that was working just fine. I can't upload the files because of github but I'll link to them on G drive. https://drive.google.com/drive/folders/0B1nweeX0k7NPcXV4WjFaTkVPNU0?usp=sharing

pfalstad, 2016-12-07:
Hmm.  I loaded that profiling output in Chrome but can't make any sense of it.  Everything's obfuscated by GWT which doesn't help.  What version of Chrome?   I assume you're using Windows 10, since you mentioned Edge?

pfalstad, 2016-12-07:
You could try running http://www.falstad.com/circuit/circuitjs2.html and see if that runs any better.  That is an older version.

modwizcode, 2016-12-07:
Yeah GWT is kinda messy imo, I'm using latest chrome ( 55.0.2883.75 m ) 64 bit. Windows 10 is the platform. The older version you linked does not have the problem.

sharpie7, 2016-12-07:
Hi Paul - in the GWT Build you could try changing the "Output Style" option to "Detailed", uploading this version and then seeing if @modwizcode or @emilfihlman can capture more logs. I am not seeing any problems either on the version on your site or mine in Chrome or FF. Maybe some odd corner case or an interaction with a plug in?

modwizcode, 2016-12-07:
I don't think it's a plugin because a clean build of chrome canary running 0 plugins has the same issue.

gitgrimbo, 2016-12-07:
I see high CPU even after clicking "Stop".  Can't really make any sense of the obfuscated code (can you enable source maps?), but seems to be a lot of time spent in `drawImage`.

Is there a tight loop calling `drawImage`?

pfalstad, 2016-12-07:
Can you try these and see where the bug was introduced:

http://www.falstad.com/circuit/circuitjs3.html
http://www.falstad.com/circuit/circuitjs4.html
http://www.falstad.com/circuit/circuitjs5.html
http://www.falstad.com/circuit/circuitjs6.html
http://www.falstad.com/circuit/circuitjs7.html

I just upgraded to the latest chrome (I was on 54 something) and suddenly I'm seeing some weird behavior.  Still not any major slowdowns, but the circuit is flickering(?), and the scope is running a little choppier.  If I go back to circuitjs3.html it works.  So something broke in circuitjs4.html maybe (and maybe also in the latest chrome)?

I can also try building a new version with more info like Ian and gitgrimbo suggested..

modwizcode, 2016-12-07:
Yeah the flickering is an early sign. If you let it keep running (or use more complicated circuits) the thing starts to freeze up and cause chrome to have to pause massively. I'm seeing it (possibly to a lesser extent) on circuitjs4.html, and it's definitely happening in circuitjs6.html. It just works very flickeringly for awhile, very laggy responses and then suddenly everything freezes.

emilfihlman, 2016-12-07:
Hi and sorry for the late response

Yeah, I'm on Chrome 55.0.2883.75 beta (64-bit) and Windows 10 and I'm also experiencing the flickering.

But, I'm also experiencing this on regular old Debian with Stable Chrome (don't have the version at this moment), with flickering, too.

Let me know if you want me to do something.

pfalstad, 2016-12-07:
ok I think i have it fixed.  could you try this:

http://www.falstad.com/circuit/circuitjs9.html

chrome 55 doesn't handle Context2d.clip() very well it seems.

sharpie7, 2016-12-07:
Ooops - I think that might be my change when I added the clip when I made the 'scope pane resizable. I do recall some oddness when I added it in but I thought I got it resolved. Perhaps my code is doing something stupid like adding lots of identical paths to the clip region or - possibly I didn't understand how to use the feature well enough. Or, of course, it could just be Chrome's fault.

modwizcode, 2016-12-07:
Yeah that change does seem to fix it. Maybe double check that codepath and if it's not obvious then post a chromium bug.

pfalstad, 2016-12-07:
Ian, your code looks fine, and it works on safari.  It's possible that you need a call to beginPath() to clear out old paths, I don't know.  For now instead of clipping, I just draw the whole circuit and then clear out the scope area.

sharpie7, 2016-12-08:
Thanks for fixing Paul. I just did a small update to allow for the "white background" option (always a confusing feature for anything to do with colours in this program!)

I will go ahead and close this issue now as this looks like a long-term solution.

closed 2016-12-08 by sharpie7