sharpie7/circuitjs1 #45: TypeError: ERROR.captureStackTrace is not a function

eigengrau, 2016-12-14:
On Firefox 50.0.2, the circuitjs version currently running at [Paut Falstad’s](http://falstad.com/circuit/) fails load any circuit, giving only the error message `TypeError: ERROR.captureStackTrace is not a function` in the web console. The menu bar and UI is shown, but the circuit canvas remains white. The error messages is locked many times a second, possibly at every simulator tick.

eigengrau, 2016-12-14:
Full backtrace from the web console:

```
getStackTrace http://falstad.com/circuit/circuitjs.html:81:7
	getOriginatingScriptUrl http://falstad.com/circuit/circuitjs.html:96:19
	trapInstanceMethod/item.obj[item.propName]</< http://falstad.com/circuit/circuitjs.html:147:28
	Zf http://falstad.com/circuit/circuitjs.html:3:34
	e$ http://falstad.com/circuit/circuitjs.html:441:22
	zU http://falstad.com/circuit/circuitjs.html:15:1948
	TA http://falstad.com/circuit/circuitjs.html:341:76
	g http://falstad.com/circuit/circuitjs.html:20:110
	Og http://falstad.com/circuit/circuitjs.html:692:27
	Rg http://falstad.com/circuit/circuitjs.html:186:44
	Qg/< http://falstad.com/circuit/circuitjs.html:164:50
	PA http://falstad.com/circuit/circuitjs.html:21:42
	<anonymous> circuitjs1-0.js
	m http://falstad.com/circuit/circuitjs9a/circuitjs1.nocache.js:9:466
	A/circuitjs1.onScriptDownloaded/< http://falstad.com/circuit/circuitjs9a/circuitjs1.nocache.js:10:56
	d http://falstad.com/circuit/circuitjs9a/circuitjs1.nocache.js:6:55
	l/e< http://falstad.com/circuit/circuitjs9a/circuitjs1.nocache.js:7:92
```

This seems to be relatively recent. I’ve been last using the deployment at Falstad’s last week or so without issues.

pfalstad, 2016-12-14:
what platform are you on (mac or windows), and what browser and browser version?

pfalstad, 2016-12-14:
you're saying it fails to even load the default LRC example?

pfalstad, 2016-12-14:
very strange.  I searched the whole source tree and didn't find any reference to captureStackTrace, getOriginatingScriptUrl, or trapInstanceMethod.  I googled for those strings and found them here: https://github.com/brave/browser-laptop/pull/1354/files#diff-7440f7b95ccc371bdd73dce0a6631831  That seems to be a browser called "brave", nothing to do with firefox.

eigengrau, 2016-12-14:
Mh, odd. I’m also seeing this on another machine running Firefox 50.0.2. Hold on, let me try an empty profile, maybe some extension is doing something weird.

eigengrau, 2016-12-14:
Okay, this was introduced by the recently updated Privacy Badger extension. Thanks for being puzzled. ;)

closed 2016-12-14 by eigengrau