sharpie7/circuitjs1 #46: Automatically decrease step size on convergence failed

emilfihlman, 2016-12-23:
It is super annoying when simulating a circuit and then having it fail and needing to start over from the beginning at a lower step size. Not to say that this process might need to be repeated multiple times, taking exponentially more time to find a functional step size.

How about instead of having a static step size we could have a dynamically changing one. For example, for every nth step, we save the circuit state and upon entering a convergence failed situation, reset to last known good state, lower step size and try again.

Now, I don't know what the situation actually is but given that we can already export the circuit at any state, it shouldn't be too hard to implement this?

pfalstad, 2021-01-07:
I should've done this 10 years ago!!

pfalstad, 2021-02-24:
done

closed 2021-02-24 by pfalstad