sharpie7/circuitjs1 #47: To slow with more parts

custume, 2017-01-01:
Hello,

I´m sorry to bother but I found the app to be way to slow with a lot of parts.

Going to drop here my designed so that you can see the lag, I don't dared add more parts and im still like 20%.

Thank you and happy new year !!!

[3.txt](https://github.com/sharpie7/circuitjs1/files/679911/3.txt)

sharpie7, 2017-01-01:
Yes, unfortunately it isn't going to work very well if you want to try and simulate something like a full ALU! It's mostly useful for demonstrating small sub-circuits for teaching etc.

You can stop the simulation and then edit the circuit without the simulation running. This should help the lag a lot.

pfalstad, 2017-01-03:
One thing you can do is to increase the time step size.  Go to Options->Other Options, and change the time step from 5u to 50u.  That will make the simulation run 10 times faster.  Maybe even 200u.  After doing this, you can reduce the simulation speed to make it less choppy.

If the time step is larger, the simulation will be less accurate for the analog part of your circuit.  But it looks like your circuit is mostly digital except for the clock generation.  So the clock frequency might be a little off if the simulation step is large, but it might be worth it.

closed 2017-05-29 by sharpie7