sharpie7/circuitjs1 #51: volt meter / scope probe showing wrong sign

crazyhog, 2017-01-16:
The +/- sign for the volt meter / scope probe seems wrong when used together with ground reference.

Please see the attached diagram.  At the positive terminal of the battery, I attached a volt meter.  At the negative terminal, I attached it to ground.   However, the volt meter reads -5v.  

[image]

I created another measurement that connect both the leads of the voltage probe.  That reads correctly 5v.  I also added a led to show the direction of the conventional current flow is correct (i.e. the battery terminal is +5v)

sharpie7, 2017-01-16:
Odd. I can't reproduce this problem - it seems to reference to ground - see this example:
http://tinyurl.com/z9sznvo

Do you have the file that displays this problem?

Now, it is an interesting question whether a meter that has one terminal unconnected should display any reading at all!

pfalstad, 2017-01-16:
The scope probe shows the voltage of the 2nd terminal minus the 1st terminal.  (The 1st terminal is where you intially place the probe, the 2nd terminal is where you drag and release it)  So I can reproduce this behavior if the 1st terminal is at ground (or empty space) and the 2nd terminal is at 5V; but it works fine if the 2nd terminal is at ground (or empty space).  So, just swap the terminals and you should be fine.

sharpie7, 2017-01-17:
DOH of course! I do think we should add a little label to differentiate the terminals on the meter though.

crazyhog, 2017-01-18:
No, the starting and ending terminal is not the issue.  The bug is caused by the y coordinate of the ending terminal somehow influencing the sign of the voltage calculation.  

Here is 2 voltage probes starting at the same corner on the circuit.  The only difference between the 2 probes is the y coordinate of their ending terminal.  You can also see this in action if you go into select and drag mode and drag the ending terminal up or down in y coordinate.  You'll see the +/- sign flip as you drag.

http://tinyurl.com/gkvwber

[image]

$ 1 0.000005 10.20027730826997 50 5 43
v 208 352 208 128 0 0 40 5 0 0 0.5
162 208 128 336 128 1 2.1024259 1 0 0 0.01
r 336 128 336 352 0 100
w 336 352 208 352 0
p 208 128 112 32 1 0
p 208 128 96 192 1 0
g 208 352 208 416 0

Sharpie, by the way, I am learning EE on my own and I really like your tool!

pfalstad, 2017-01-18:
Oh, I forgot about that.  I apparently did that on purpose, in an attempt to guess which way the voltmeter should be oriented.  Higher on the screen is generally higher voltage, so we assume the top end of the voltmeter should be positive.  It's a feature!

crazyhog, 2017-01-18:
Ugh, I really cannot agree with that as a feature :(  

Actually, I agree with both you and sharpie in your earlier comments.  

The rules are:
1.  v = terminal 2 - terminal 1 (or vise versa is fine; consistent is key)
2.  label the probe point and the reference point on the screen.  I think its a very good idea to make this work in a conceptual way that is similar to a real oscilloscopes probe with a grounding reference clip.  If the ground is not specified assume it's the same as circuit ground.  

One more suggestion, the tool has a feature to show a component in the scope.  In this situation, the tool is doing differential probing.  Again, labeling the probing point and reference point would be very nice.

sharpie7, 2017-01-19:
Yes, I was playing with the the sim last night and I realized that the scope probe had a mind of its own as to which end was positive! 

I do think this behaviour needs a revision. It's pretty odd.

sharpie7, 2017-01-22:
Should now be fixed on mine (also another polarity thing to do with no polarity being shown on AC sources with a bias). Thanks for reporting.

closed 2017-01-22 by sharpie7