sharpie7/circuitjs1 #52: Feature request: Allow for setting of resistance for MOSFETS and inductors

sterna, 2017-01-20:
I'm trying to simulate a high-power buck converter, but realized that the MOSFETs have a very high on-resistance (I measured it to about 200ohm). The inductors I don't know the resistance of. It would be nice it the resistance value is at least documented, and at best, possible to set.

pfalstad, 2017-01-20:
Hi,

For inductors, you can just put a resistor in series with it for now.

For MOSFETs, the resistance depends on current, of course.  You can specify the MOSFET curve by setting the threshold and the beta.  Beta is equal to 1/(RdsON * (Vgs-Vt)), so if you know the threshold voltage, and the Vgs voltage you used to measure RdsON, you should be able to calculate beta.

pfalstad, 2017-01-20:
Just to be clear, the inductors in the simulator don't have any resistance.  So to simulate a real inductor you need to put a resistor in series.

closed 2017-05-29 by sharpie7