sharpie7/circuitjs1 #55: Cannot create text with a value of +

indymaynard, 2017-01-25:
Before I get to the point of this, I want to say that this project is incredible! I am an instructor for a theory course, and this project has inspired me to create a handout using the capabilities of this excellent project.

The only issue I am having is adding a "text box" with a value of just a plus sign. I can create it, and it saves in the code for the circuit, but will not load when pasting the code into the app, and even disappears from the code when trying to copy it again.

My course teaches electron flow, so it's important when I create a current source that I label the direction of electron flow, since the conventional flow symbol for current points in the opposite direction.

Thanks for the outstanding efforts put into this app.

pfalstad, 2017-01-25:
Hmm.  Yeah that's a bug.  We should probably use escape()/unescape() (from CustomLogicModel.java) for text elements.

sharpie7, 2017-01-25:
Good catch - should now be working on my version.

Didn't use escape()/unescape() in the end as the problem is really just "+" characters which are both allowed in the text and are also delimiters for tokens in the export format.

closed 2017-01-25 by sharpie7