sharpie7/circuitjs1 #62: High voltage circuits cause convergence failing on any time steps

emilfihlman, 2017-02-04:
I tried to generate high voltage pulses with the following circuit (and also tried with a resistor in the capacitors place)
$ 1 1e-9 0.10425469051899915 50 5 43
R 464 320 384 320 1 2 1000 2.5 2.5 0 0.5
f 464 320 512 320 16 1.5 0.02
g 512 336 512 400 0
l 416 240 512 240 2 0.01 0.009968548165247795
R 416 240 352 240 0 0 40 5 0 0 0.5
187 720 240 720 352 0 1000 1000000000 1000 0.001
g 720 352 720 400 0
w 512 240 576 240 0
w 576 240 720 240 0
c 512 240 512 304 2 0.000009999999999999999 3.846836422686465
o 5 64 0 2083 10 0.00009765625 0 -1 0

However, going down to 1n time step, it still fails to converge.

pfalstad, 2017-02-04:
Sorry about that, I have a fix for this which will be out soon.

emilfihlman, 2017-02-17:
Has there been progress made regarding this issue?

pfalstad, 2017-02-18:
whoops, I thought I released this fix already!  I'll bring it out soon.

pfalstad, 2017-02-18:
this is fixed in 2.0.1, hit reload on my site and it should work.

sharpie7, 2017-02-26:
Fixed.

closed 2017-02-26 by sharpie7