sharpie7/circuitjs1 #64: Error in circuit explanation for astable multivibrator

yacsaw, 2017-02-13:
On page www.falstad.com/circuit/e-multivib-a.html the circuit explanation appears to have an error.  First it says that the Q1 collector current discharges C1, then in the next paragraph it says that the collector current charges C1, which I believe is correct.

pfalstad, 2017-02-13:
Well, the current discharges C1 to neutral and then starts to charge it slightly in the other direction.  I changed the second paragraph to be less confusing.  See what you think.

yacsaw, 2017-02-15:
Mr. Falstad,

Thank you for responding to my concern.  I appreciate that it is challenging to explain the operation of the multivibrator in just a few words, and I am not the best one to advise you since I am just in the process of learning about it.  I think I see what was confusing me.  While Q1 is conducting as described in the first paragraph, current from the left 1K resistor flowing to the left through C1 discharges it because initially its right side is negative compared with its left side.  But if you don’t know which way C1 is charged, seeing current flow to the left through it as part of a path from +5V to ground looks like it would charge it up (at least it can look that way to a beginner like me).  So for improved clarity maybe the first paragraph could mention which side of C1 is positive at the start of Q1’s conducting phase.

Yours truly,
Steve Wise

sharpie7, 2017-02-26:
Closed as Paul has updated his site. 

The circuit explanations are not actually part of this distribution!

closed 2017-02-26 by sharpie7