sharpie7/circuitjs1 #65: Sequential Switching Bistable Multivibrator

Lemtil, 2017-02-13:
I had trouble getting this circuit working.
The problem ended up being the transistors being exactly identical, which is a very bizarre problem to have. Would it be hard to implement component value errors?

$ 1 0.000005 0.13386567243530942 50 5 50
R 176 128 176 80 0 0 40 5 0 0 0.5
r 176 128 176 192 0 3900
r 400 128 400 192 0 3900
w 176 128 400 128 0
w 176 192 208 192 0
w 400 192 368 192 0
r 208 192 256 192 0 47000
r 320 192 368 192 0 47000
w 256 192 320 224 0
w 320 192 256 224 0
w 176 192 176 208 0
w 400 192 400 208 0
t 368 224 400 224 0 1 -4.073909850738368 0.32816588861736995 101
t 208 224 176 224 0 1 -6.3375382159237965 -1.808826491257624 100
w 368 224 336 224 0
w 336 224 320 224 0
w 256 224 240 224 0
w 240 224 224 224 0
w 224 224 208 224 0
d 320 224 320 256 1 0.805904783
d 256 224 256 256 1 0.805904783
r 336 224 336 272 0 100000
r 224 224 224 272 0 100000
w 400 240 400 272 0
w 400 272 336 272 0
w 336 272 224 272 0
w 224 272 176 272 0
w 176 272 176 240 0
w 256 320 256 256 0
w 320 320 320 256 0
c 256 320 288 320 0 1e-9 2.605348736966969
c 320 320 288 320 0 1e-9 4.7562609050011115
w 288 320 288 352 0
w 288 352 112 352 0
R 112 352 112 320 0 2 2000 5 0 0 0.5
w 176 192 144 192 0
w 400 272 464 272 0
R 464 272 464 304 0 0 40 0 0 0 0.5
r 432 192 432 240 0 220000
r 144 192 144 240 0 220000
w 400 192 432 192 0
O 432 192 480 192 0
w 432 240 432 320 0
w 432 320 320 320 0
w 256 320 144 320 0
w 144 240 144 320 0
o 34 1 0 2091 5 0.00078125 0 -1 0
o 41 1 0 2090 5 0.00009765625 0 -1 0

sharpie7, 2017-02-13:
Thanks for sending an interesting circuit to look at.

I have seen real-world situations where bistables like this can get stuck in odd states too. I think that as a circit designer it is better practice to deliberately make the two sides slightly asymmetrical (eg by changing one of the resistor values) to force one to "lead" rather than relying on the random variation in the transistor parameters - though in the real world that would probably work.

Here is my alternative:
http://tinyurl.com/gsvyj5j

From my point of view it is better that this simulator behaves predictably - ie simulates to the best of its ability the components specified.

pfalstad, 2017-02-13:
I had the same problem when creating the astable multivibrator example in the circuits menu.  I had to change one of the resistor values to make it asymmetrical.  I agree that the simulator should be predictable.  Some circuits, like current mirrors, require identical components.

Lemtil, 2017-02-13:
Thanks for the schooling sharpie7, one more design philosophy in the old toolbelt.

http://tinyurl.com/jfvzuas

closed 2017-06-08 by sharpie7