sharpie7/circuitjs1 #66: Feature

jmoyni, 2017-02-15:


jmoyni, 2017-02-15:
I was just wondering if you could upload a downloadable version of this applet so that I can run it as a standalone application on my Windows PC. I am aware that I could just download the entire page as an HTML file for offline use, but a standalone desktop app would look much slicker and would be AWESOME as an EE student. Your applet is a better circuit simulator than any others out there in my opinion, and would love to actually have it on my computer.

yoflippo, 2017-03-13:
I agree, this would be a great feature. I happen to live in an area with unreliable internet. This would be a great feature just like the old Java version of the circuit simulator.

sharpie7, 2017-05-28:
Offline use is not supported and I do not intend to work on addressing it. I have added an explantion of why under "Offline Use" here: http://lushprojects.com/circuitjs/

closed 2017-05-28 by sharpie7