sharpie7/circuitjs1 #67: Certain small circuit causes extreme performance issues

dudejuice, 2017-02-16:
I've noticed that when I build a simple square wave inverter, the applet hiccups profoundly whenever the voltage reverses direction, and will under some circumstances slow to several seconds per frame.

I've reproduced this on the JS and Java versions in both Firefox, Chromium, and local Java. Below is the circuit in linkified form. I don't know that much about discrete electronics, but I have a lot of fun messing around in this rad little applet.

http://tinyurl.com/zgwxeat

Lemtil, 2017-02-17:
It doesn't like inductive spikes, or infinite feedback loops.

In the meantime. Here's a circuit that uses the same principles, but has smoothing components, and more sane values.
http://tinyurl.com/hk2dyoa

dudejuice, 2017-02-19:
Thanks for responding and putting together that circuit. That's really unfortunate. I enjoy spikes and non-sane values.

I've noticed a lot of weird transformer and diode-related performance issues.

Lemtil, 2017-02-19:
Things get weird because of the lack of parasitics, mostly. Make sure you add in series resistors, parallel capacitors, reduce transformers to milli/micro inductance values, and then you can do pretty much whatever you want.

Like, go nuts.
http://tinyurl.com/z3swc7j

dudejuice, 2017-02-19:
Oh my god it's hideous! Thanks, I'll do some experimenting and see what I come up with. My goal is to make an inverter that you can run a load like the lamp componant on.

pfalstad, 2021-12-11:
This seems to be fixed now.

closed 2021-12-11 by pfalstad