sharpie7/circuitjs1 #68: Feature Request: a couple node editing features

EmergencyTemporalShift, 2017-02-17:
First of all, the project is awesome and is probably the best free GUI circuit simulator out there. That said, there are a few editing modifications I feel should be added:

1. There should be an option to flip components (swap nodes).
This allows you to fix the scope probe being upside down without having to flip the components manually. (Obviously, this would be for resistors, capacitors, inductors, LEDs, etc.)
2. There should be a drag option for all nodes under the cursor.
Repositioning nodes in this mode would move all the nodes under the cursor, repositioning components would just move all of their nodes. (Integrated chips would not work with this.)
3. There should be an interact-only mode.
In this mode, you could toggle switches or potentiometers, but not accidently drag components around. This might be helpful if you have a specific circuit you want to simulate and don't want to change it (Like maybe a classroom setting).

sharpie7, 2017-02-26:
Flip added in recent version. Title changed to reflect the other points.

EmergencyTemporalShift, 2017-02-26:
Nice, thank you.

pfalstad, 2019-03-07:
Thinking about the interact-only mode..  Should this be saved and loaded with each circuit, like Show Current / Show Voltage?  Or should it be global, and possibly saved in local storage, like European Resistors?

I'm concerned about someone setting it by accident and not being able to edit their circuit..  But I could probably just have a popup error message if someone tries to select a component, maybe.

EmergencyTemporalShift, 2020-12-05:
I moved the two unimplemented ones and I'm marking the completed one as closed.

closed 2020-12-05 by EmergencyTemporalShift