sharpie7/circuitjs1 #70: Features - components variable resistors, not potentiometers, non-linear variable resistors

alanwevans, 2017-02-21:
I can use pots to simulate variable resistors by tying the ends together, but it would be nice to have a simpler device for variable resistors.

While we're at it, how about things like LDRs, PTC/NTC thermistors?  They could be linear approximation or perhaps have inputs for approximate curves.  Maybe use B parameter?  Some datasheets seem to provide it.

https://en.wikipedia.org/wiki/Thermistor#B_or_.CE.B2_parameter_equation
http://reprap.org/wiki/MeasuringThermistorBeta

Or just point entry to approximate a curve.  Take 5 - 8 points like these.  There might even be some kind of java library for fitting curves to a set of points.
http://www.bapihvac.com/wp-content/uploads/2010/11/Thermistor_100K.pdf

Thanks,
-Alan

billcollis, 2017-02-21:
LDRElm.java ID=374

package com.lushprojects.circuitjs1.client;

import com.google.gwt.user.client.ui.Label;
import com.google.gwt.user.client.Command;
import com.google.gwt.canvas.dom.client.CanvasGradient;
import com.google.gwt.event.dom.client.MouseWheelEvent;
import com.google.gwt.event.dom.client.MouseWheelHandler;

/*Bill Collis - June 2015  */

class LDRElm extends CircuitElm implements Command, MouseWheelHandler {
	double position; //of the slider 0.005 to 0.995
	double resistance; //based upon slider position
	double minLux, maxLux; 
	double lux; 

	Scrollbar slider; 
	Label label;
	String sliderText;
	
	//constructor - when initially created
	public LDRElm(int xx, int yy) {
		super(xx, yy);
		//setup();
		minLux = 0.1; //dark
		maxLux = 10000; // sunlight
		position = .34; 

		lux = LuxFromSliderPos();
		resistance = calcResistance(lux); 
		sliderText = "Dark--------------------Bright";
		createSlider();
	}
	
    //constructor - when read in from file
    public LDRElm(int xa, int ya, int xb, int yb, int f,
    		StringTokenizer st) {
    	super(xa, ya, xb, yb, f);
		minLux = 0.1; //dark
		maxLux = 10000; // sunlight
		position = new Double(st.nextToken()).doubleValue();
		lux = LuxFromSliderPos();
		resistance = calcResistance(lux); 
		sliderText = st.nextToken() ;
    	while (st.hasMoreTokens())
    		sliderText += ' ' + st.nextToken();
    	createSlider(); //uses position to set the slider   
   }
    
    //void setup() {
    //}
    
    int getPostCount() { return 2; }
    int getDumpType() { return 374; } //LDR
    
    Point getPost(int n) {
	return (n == 0) ? point1 :  point2;
    }
    
    //data for file saving - make sure it matches order of items in file input constructor
    String dump() { 
    	return super.dump() + " " + position  + " " + sliderText; 
    	}
    
    void createSlider() {
    	sim.addWidgetToVerticalPanel(label = new Label(sliderText));
    	label.addStyleName("topSpace");
    	int value = (int) (position*100);
[... 123 lines cut ...]
    
    public void onMouseWheel(MouseWheelEvent e) {
    	if (slider!=null)
    		slider.onMouseWheel(e);
    }

	double calcResistance(double lux) //knowing the lux
    {
		//double loglux = Math.log10(lux);
		//double slope = -1.4;
		//double intercept = 7.1;
		//double logR = 	(loglux-intercept)/slope;
		
		//return Math.round(Math.pow(10, logR));
		double r = (maxLux-lux+1)*10;
		
		r = Math.round(r);
		return r;
    }
	double LuxFromSliderPos() //knowing slider position etc
	{
		return maxLux * position + minLux ;
	}

}

billcollis, 2017-02-21:
ThernistorElm.java ID=350

package com.lushprojects.circuitjs1.client;

import com.google.gwt.user.client.ui.Label;
import com.google.gwt.user.client.Command;
import com.google.gwt.canvas.dom.client.CanvasGradient;
import com.google.gwt.event.dom.client.MouseWheelEvent;
import com.google.gwt.event.dom.client.MouseWheelHandler;

/*Bill Collis - June 2015 
 * datasheet for Vishay thermistor  ntcle100
 * thermistor model
 *  view-source:http://www.giangrandi.ch/electronics/ntc/ntc.shtml
 *  http://www.giangrandi.ch/electronics/ntc/ntcparam.html
 *  no heating effects
 *  ID = 350
 *  add id to CirSim.constructElement and part to CirSim.createCe and to CirSimcomposeMainMenu
 */

class ThermistorNTCElm extends CircuitElm implements Command, MouseWheelHandler {
	double position; //of the slider 0.005 to 0.995
	double resistance; //based upon slider position
	double minTempr, maxTempr; //Celsius - note min is -40, max is +150 degC 
	double temperature; //calculated from slider value (0.005 - 0.995) ratios of minTempr - maxTempr
	double r25, r50; //the values that a user can input from a datsheet r 225 degc and r at 50degC
	double rneg40; //maximum resistance - will be at -40 degC 
	double b25100; // constant based upon 2 values of R for 2 temperatures
	double t0 = 273.15;
	double t25 = t0 + 25;

	Scrollbar slider; //from Pot
	Label label;
	String sliderText;
	
	//constructor - when initially created
	public ThermistorNTCElm(int xx, int yy) {
		super(xx, yy);
		//setup();
		minTempr = -40;//celsius
		maxTempr = 150; 
		r25 = 10000; //default 10k thermistor e.g. NTCLE100E3010 Vishay
		r50 = 3605;
		position = .34; //25 degC for -40 to 150 degC
		//thermistor calcs
		rneg40 = calcResistance(minTempr); //for 10k ntc about 400k	
		b25100 = calcB25100(); //	
		temperature = temprFromSliderPos();
		resistance = calcResistance(temperature); 
		sliderText = "Temperature -40 150 C";
		createSlider();
	}
	
    //constructor - when read in from file
    public ThermistorNTCElm(int xa, int ya, int xb, int yb, int f,
    		StringTokenizer st) {
    	super(xa, ya, xb, yb, f);
    	r25 = new Double(st.nextToken()).doubleValue();
    	r50 = new Double(st.nextToken()).doubleValue();
    	minTempr = new Double(st.nextToken()).doubleValue();
    	maxTempr = new Double(st.nextToken()).doubleValue();
    	position = new Double(st.nextToken()).doubleValue();
		//thermistor calcs
		rneg40 = calcResistance(minTempr); //for 10k ntc about 400k	
		b25100 = calcB25100(); //
		temperature = temprFromSliderPos();
		resistance = calcResistance(temperature); 
		sliderText = st.nextToken() ;
    	while (st.hasMoreTokens())
    		sliderText += ' ' + st.nextToken();
[... 157 lines cut ...]
    }
    
    public void onMouseWheel(MouseWheelEvent e) {
    	if (slider!=null)
    		slider.onMouseWheel(e);
    }

	double calcResistance(double tempr) //knowing the temperature
    {
		return Math.round(r25 * Math.exp(b25100 * ((1 / (tempr + t0)) - (1 / t25))));
    }
	double temprFromSliderPos() //knowing slider position etc
	{
		return Math.round( position * (maxTempr - minTempr) + minTempr);
	}
	//determine constant B25100 - when knowing two R values at two temperatures
	double calcB25100() //given R25=10000 and R50=3605  B25100 will be 3932
	{
		double kelvin1 = t0 + 25;
	    double kelvin2 = t0 + 50;
	    return ( Math.log(r25) - Math.log(r50) ) / ( (1/kelvin1) - (1/kelvin2) );
		
	}
	
}

billcollis, 2017-02-21:
I have pulled these two from the ones I have built and am testing. They are free to be added to  Iain and Pauls work, They do need Iain and Paul's copyright info put back into them and testing as I have stripped off some extra features I have in the version of the simulator I am working on

pfalstad, 2017-02-25:
Thanks Bill!  I will take a look at these when I get a chance.

mrfrenzy, 2017-06-06:
Thankyou Paul for the very useful simulator and thaknyou Bill for the additions.
Do you think it would be possible to create a Varistor aswell?

Regards,
Pierre

pfalstad, 2020-12-12:
Ok I finally took a look at these and am incorporating them now.  Thanks Bill!

ipatrol, 2021-12-04:
How about adding optional heating effects to the thermistor? I'd suggest starting with the lamp model for that, and allow both PTC and NTC forms by simply allowing the TC to be positive or negative.

pfalstad, 2021-12-11:
Ok the main request here is done but someone needs to go through all the comments and make separate issues for all the additional requests.  🙂