sharpie7/circuitjs1 #72: Power and common pins for components.

Pau563, 2017-02-22:
Is it possible to get access for the power pins and other pins. Example circuits with different voltages to components. 
In the image using a 7 segment display I had to work around not having access to the common Anode or common Cathodes.  The only way to actually get it work to what I needed was to use Individual LED's.

[image]

EmergencyTemporalShift, 2017-02-24:
I would recommend either a global or per component toggle so you don't have to deal with them if you don't want to. Also forward compatibility. Unrelated but, a custom segment display I think has been done before somewhere, you need to upload a binary image.

pfalstad, 2019-03-06:
added in ddd9487663f5d2a302ea9ee1f01f90892ee33c9c (for 7 segment display)

Pau563, 2019-03-06:
> added in ddd9487 [link: https://github.
> com/sharpie7/circuitjs1/commit/ddd9487663f5d2a302ea9ee1f01f90892ee33c9c]
> auth/AYwmPo1Vm8aXvSt465YQ23mroiAQJ21Oks5vTwv-gaJpZM4MI-8b].[Image]

Thank You.

closed 2019-03-06 by sharpie7