sharpie7/circuitjs1 #79: Bug: Text doesn't snap to grid. Moving selections with text and components creates strange results.

sharpie7, 2017-02-26:
Text doesn't snap to grid - no strong opinion on that though it may be seen as a little inconsistent.

However if you make a selection that includes text and components in the same selection then dragging can get very odd as components now move off the grid.

pfalstad, 2017-02-26:
Text doesn't snap to grid on purpose, so you have more flexibility in positioning it.

Definitely if there's a selection that includes non-text, we should snap to grid.  We try to do this but not correctly, which results in very strange behavior.  This only happens if the mouse is positioned over the text element when dragging.

sharpie7, 2017-03-04:
Fixed in my version

closed 2017-03-04 by sharpie7