sharpie7/circuitjs1 #80: Exported files won't open - Maybe Custom Digital Chips at fault

TomWhitwell, 2017-02-26:
LOVE this tool, use it all the time, thank you so much for your hard work.  
I've had trouble re-opening saved files - I get this error: 

Error on line 1 of model description

I wonder if it's because I'm using a Custom digital chip in this model. 

A few examples: 
http://tinyurl.com/z9lsdty
http://tinyurl.com/jqxapkj

pfalstad, 2017-02-26:
Does this happen with "Export as Text" or "Export as Local File"?  I can only get it to happen with "Export as Link".

TomWhitwell, 2017-02-27:
Aha, yes, I just tested my custom chip alone. Export as link failed, export as text worked.

pfalstad, 2017-02-27:
Apparently "=" characters are the problem.  Working on a fix.  It seems that = in text elements don't work either.

sharpie7, 2017-05-28:
Fixed - I believe.

closed 2017-05-28 by sharpie7