sharpie7/circuitjs1 #81: Feature - instructions on compiling as Java or other stand-alone app

SeeJayDee, 2017-03-02:
Hi,

I usually run the original java applet (v1.6i) from my computer, as it's more portable and works despite unreliable internet (hooray Australia). However, I've noticed that all of the new parts, features and examples are only being added to the new Javascript version of the app. 
Which is 100% understandable.

But I'd like to know if there was any way to insert those new features into the Java version source code & compile it?

Cheers,
SeeJayDee

pfalstad, 2017-03-02:
You'd have to port the changes back to java, which would be a fair amount of work.  New components and changes to components would be pretty easy to port back.  UI changes would be a lot harder.

To run offline, the simplest thing would be to install Eclipse and build the app from source.  Instructions here: https://github.com/sharpie7/circuitjs1

sharpie7, 2017-03-02:
Yes, I agree with Paul 100%

How bad is using the web app from where you are? Once the page is loaded there shouldn't be any more interaction with the server unless you load a new example circuit, use the Dropbox integration or use the URL shortener.

Harvie, 2017-03-28:
Is it possible to store HTML/javascript files localy and open it in the browser directly from disk? I think at least this option should be supported.

sharpie7, 2017-05-29:
Offline use is not supported because of problems inherent in the software platform the project uses. I have added an explantion of why under "Offline Use" here: http://lushprojects.com/circuitjs/

Though I do not intend to work on offline use I have created an issue #121 which can be used by people who want to work on this themselves to discuss ideas and progress.

closed 2017-05-29 by sharpie7

Harvie, 2017-05-29:
So i suppose as workaround you can localy install webserver like lighttpd, thttpd or apache and use it to serve circuitjs.