sharpie7/circuitjs1 #86: Feature - Math operations in scope

easyas314159, 2017-03-04:
It would be rather useful in some instances to be able to perform math operations (+, -, *, /) on signals in the scope. Either using some extension of 'combine', or by adding the operators as Voltage based helper components that could be added to the scope. In the case of Voltage based helper components, a component to convert current directly to a Voltage would also be useful.

pfalstad, 2017-03-04:
I'm working on a component that lets you perform math operations on its inputs, so once that is ready you could use that to generate values that could be scoped.

pfalstad, 2019-03-06:
ok now you can use a VCVS (or CCVS) to generate voltages which you can then plot in a scope.

closed 2021-12-10 by pfalstad