sharpie7/circuitjs1 #89: Lamp problems

reitho, 2017-03-06:
Hello,

I am using your great tool in an austrian vocational school for electrical engineering. For a short time, when using the lamp (beta), I get a _nan/infinite matrix!_: [http://tinyurl.com/jjcmp86]...

Thank you so much for your support!

Kind regards,
Thomas Reitinger

pfalstad, 2017-03-06:
Thanks, I have a fix for this which will be released soon.

BTW, if you want a German version of the simulator and are willing to translate strings, let me know.

reitho, 2017-03-06:
Thanks for fast response :) So I will wait for the fix...

German version would be so cool. Maybe you can remember that we already had a conversation about that. I really would be happy to work on a german version. Just send how you need the data...

Bye, Thomas

pfalstad, 2017-03-06:
Thanks, I sent you an email about that.  I hope your email address hasn't changed since 2014.  :)

sharpie7, 2017-06-08:
Now appears to be fixed!

closed 2017-06-08 by sharpie7