sharpie7/circuitjs1 #91: "BUG" MOSFET body diode and gate not working properly

Jaydee500, 2017-03-16:
I love this simulator but, I have found that the MOSFET body diode does not work.
If I reverse bias a MOSFET it won't conduct if the gate is floating. Strangely if i then connect the gate to the source it starts to conduct even though in this configuration VGS = 0V. It seems it is referencing a positive gate voltage to switch on but it should be gate voltage > Source voltage for switch on.

Link below

http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Af+304+240+352+240+0+1.5+0.02%0Aw+352+256+352+304+0%0Aw+352+224+352+176+1%0A172+416+304+416+272+0+6+5+5+0+0+0.5+Drain+Voltage%0Ag+464+176+464+192+0%0Aw+352+304+416+304+0%0Aw+464+176+352+176+0%0As+352+304+256+304+0+1+false%0Aw+256+304+256+240+0%0Aw+256+240+304+240+0%0Ao+0+64+0+4099+5+0.2+0+2+0+3%0A

pfalstad, 2017-03-16:
That is true, the body diode is not implemented.  In this case the gate terminal is considered to be unconnected, so the simulator ties it to ground.

With the switch turned on, the MOSFET acts like the drain and source are swapped.  Since there's no body diode, the MOSFET is symmetrical.  So the actual Vgs = 5V (using the drain as the source).

Jaydee500, 2017-03-16:
Hi Thanks for the quick response.
Since the simulated behavior does not match the circuit is this something you can correct for?
Can the body diode be implemented or do you consider this as a parasitic element of the component that users should add themselves.

Do all floating terminations get tied to ground?

Why would the terminals get swapped in behavior when the gate was shorted to the source. If the body diode is not in circuit then the FET should be switched off.

Thanks

Thanks

pfalstad, 2017-03-16:
Yes, adding a diode would be easy enough.  It would be nice to know more specific parameters of the diode, but I suppose any diode would be better than nothing.

Sorry, I should have said that there's no body in the model we're using (source is not tied to body), so the MOSFET is treated as symmetrical.  If we added a diode, it obviously wouldn't be.

closed 2017-05-29 by sharpie7