sharpie7/circuitjs1 #93: Invert signal on 'scope

ThePicMan, 2017-03-26:
Hi,
I'd like to ask for a very basic feature which I wished to see implemented since many years ago.. but it's not there.

It's just an option to invert the Y axis on scope signals, as sometimes I prefer to see the signal "negated".

Thank you very much and please keep the great work.

--
TPM

pfalstad, 2017-03-26:
You could negate a signal by attaching the negative terminal of a voltmeter to the signal, with the positive tied to ground, like so:

http://tinyurl.com/lus5nhj

ThePicMan, 2017-03-26:
Cool, thanks.

ThePicMan, 2017-04-03:
One more question, but if it's not allowed here please let me know, I will open a new thread.

In the voltage generator, if I choose Pulse waveform there doesn't seem a possibility to choose the pulse width. I have simulated the arrangement via a square wave and duty cycle, but it would be nice if (unless it's already there, but I can't see it) there was (for the pulse section) also the pulse width parameter (in degrees or %), together with the others that are already supported (DC offset, phase offset, frequency and peak voltage).

Thank you very much for your excellent simulator.

Kind regards,
TLM

pfalstad, 2017-04-03:
A pulse waveform is just the same as a square wave with a short fixed duty
cycle. I'm not sure why it's even there.. maybe I added it before I made
the duty cycle adjustable on the square wave.

ThePicMan, 2017-04-04:
The pulse waveform is still useful, because it's referenced to 0 volts. The square one instead spans +/-V.

I needed to produce two pulses, first one negative and one positive. I have put two such generators in serie (with the necessary phase offset) and voilà. I've set 20V for one, and -20V for the other, and I get a nice tra of pulses, first -20V then 20V. The only limitation is that I couldn't set the pulse width.

So I *had* to switch to the square wave generators, and to make them equivalent I must specify half voltage (10V) and the rest (the other 10V) must go into DC offset. This is not intutive.

So, if you care, I still request to add the pulse width (in degree or %, better the former if it can be only one) to the pulse generator, and absolutely not to remove this feature, which is handy and intuitive. Why use a square wave generator with half the wanted voltage that must be moved to the DC offset, when you really want to generate a pulse of x degrees width anyway?

With kind regards,
TPM

pfalstad, 2017-04-16:
ok pulse waveforms have adjustable duty cycle now.

ThePicMan, 2017-12-27:
Thank you, thumbs up!

sharpie7, 2019-03-06:
Fixed.

closed 2019-03-06 by sharpie7