sharpie7/circuitjs1 #94: Add options to change colors for colour blind

kaushalpkk, 2017-03-29:
Add options to change colors for colour blind, the greens and reds in the current flow also make a problem.

sharpie7, 2017-06-08:
Thanks for the suggesting.

A way to do this without adding a lot of complexity to the program would be to add an option to allow the user to swap one of the red/green defaults to blue (I am thinking of allowing swapping green to blue). This would avoid the most common problem of red/green colour blindness and should allow people with some colour vision to choose an option they should be able to distinguish.

I have tried doing a green/blue swap on a test version. It looks kind-of funky - a bit like blue-LED lit gadgets.

If you feel this is a reasonable way to go then I will do an update.

sharpie7, 2017-06-10:
Feature added to latest version - just uploaded to my site.

closed 2017-06-10 by sharpie7