sharpie7/circuitjs1 #96: Diodes not working as expected

SlowBro904, 2017-03-31:
Turn on either switch and both wires show greater than 3V. I wouldn't think that would work in real life, but then I'm new at this so I could be very wrong.

http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Ad+416+208+464+208+1+0.805904783%0Ad+416+304+464+304+1+0.805904783%0Aw+464+208+464+304+2%0As+304+304+368+304+0+1+false%0Ar+304+208+304+160+0+1000%0As+304+208+368+208+0+1+false%0AR+304+160+304+128+0+0+40+5+0+0+0.5%0Aw+368+208+416+208+2%0Aw+368+304+416+304+2%0Ar+464+304+464+352+0+10000%0Ag+464+352+464+368+0%0Aw+304+304+304+208+0%0A

darconeous, 2017-03-31:
I think the simulator is working just fine in this case. That 10k resistor is what is preventing the voltage drop from being very significant. [Lowing it to 1k](http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Ad+416+208+464+208+1+0.805904783%0Ad+416+304+464+304+1+0.805904783%0Aw+464+208+464+304+2%0As+304+304+368+304+0+0+false%0Ar+304+208+304+160+0+1000%0As+304+208+368+208+0+0+false%0AR+304+160+304+128+0+0+40+5+0+0+0.5%0Aw+368+208+416+208+2%0Aw+368+304+416+304+2%0Ar+464+304+464+352+0+1000%0Ag+464+352+464+368+0%0Aw+304+304+304+208+0%0A) will give you a voltage drop under 3v.

SlowBro904, 2017-03-31:
How is it that both diodes show voltage on both switch sides when only one is turned on? Shouldn't it be that no voltage is passed through?

darconeous, 2017-03-31:
Since the switch is a perfect switch, the left side of the disconnected diode is effectively floating.  Diodes have a very high impedance in the reverse direction, but it isn't zero—so that leakage is pulling up the voltage of the left side of the diode. It shows 4v, but you if you put [even the smallest load on it it would drop to zero](http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Ad+416+208+464+208+1+0.805904783%0Ad+416+304+464+304+1+0.805904783%0Aw+464+208+464+304+2%0As+304+304+368+304+0+1+false%0Ar+304+208+304+160+0+1000%0As+304+208+368+208+0+0+false%0AR+304+160+304+128+0+0+40+5+0+0+0.5%0Aw+368+208+416+208+2%0Aw+368+304+416+304+2%0Ar+464+304+464+352+0+10000%0Ag+464+352+464+368+0%0Aw+304+304+304+208+0%0Ar+416+304+416+352+0+10000000000%0Aw+416+352+464+352+0%0A).

SlowBro904, 2017-03-31:
Ok, so newbie error. Thanks!!

closed 2017-03-31 by SlowBro904