sharpie7/circuitjs1 #100: some error like:"Matrix error"

wangfei1233, 2017-04-17:
hi,dear pfalstad :
    glad to chat with you,I am a student.I found a Error,if we link the circuit like: http://tinyurl.com/munl2qj
, will throw a error "Matrix error".  It likes that if we short circuit a device which is 0 resistance(switch,wire,Ammeter),this error will appear.
   then,thank you very much.My English is bad.

OmegaHaxors, 2017-05-08:
It's because you have a wire loop. The simulator doesn't like those and messes with all the math. If you want to simulate a short, consider using an SPDT switch or at the very least use a tiny bit of resistance.

wangfei1233, 2017-05-08:
yes.I had used a tiny bit of resistance.Now it is wonderful ~ thanks~

sharpie7, 2017-05-28:
Behaviour is as intended.

closed 2017-05-28 by sharpie7