sharpie7/circuitjs1 #107: feature: JFET forward-biased

Felthry, 2017-05-21:
Currently, the simulation errors if you try to forward-bias the gate-source diode of a JFET. In reality, this would result in diode-like behavior, plus the parasitic resistance of the body semiconductor. I feel like it would be better to simulate this behavior, even if it's almost always unintended, than to simply throw an error and halt.

pfalstad, 2017-05-22:
The issue here is that I have had trouble finding specifics on what the correct behavior should be in that case.

Felthry, 2017-05-22:
Have you tried looking at SPICE models?
I would imagine it would function as a diode of some sort, though I don't know what sort of forward voltage it would have. The channel resistance would come into play too, but that is ideally zero.

pfalstad, 2017-05-22:
SPICE models tend to be much too complicated to translate into code.  But I could certainly run some test circuits in SPICE and see what sort of behavior I get.

Felthry, 2017-05-22:
[This](http://edadocs.software.keysight.com/pages/viewpage.action?pageId=5926333) page and [this](http://www.ee.bgu.ac.il/~circlab/experiments/JFETmodel.pdf) document might be helpful. Both contain general-purpose models of JFETs, for forward- and reverse-biased modes. It also might be useful to ask around on e.g. https://electronics.stackexchange.com or a similar website for someone with more experience with JFET modelling.

sharpie7, 2017-05-28:
Fixed (I believe!)

closed 2017-05-28 by sharpie7