sharpie7/circuitjs1 #108: feature: A few more examples

Felthry, 2017-05-21:
Apologies; I know this is not the correct place to put this but I can't seem to find what _is_.

I have a collection of circuits that I made in the old Java version, some of which may make for useful examples (for example, I fixed the existing subranging ADC, which had and still has a minor error, I made several chaotic oscillator circuits in the same vein as the existing Rossler circuit, there's a couple interesting demonstrations of the lambda diode, an oscillator made with relays, and so on). I am not good at git, and in the interest of not making a complete mess of things I would like to just send the files to someone who is already familiar with this project, for possible inclusion. How would be a good way to go about doing this?

pfalstad, 2017-05-21:
hi, you can mail them to me, java@falstad.com

Felthry, 2017-05-21:
Just realized I should probably note here that I did so, for others who might be curious about the status of this.

sharpie7, 2017-05-28:
Included.

closed 2017-05-28 by sharpie7