sharpie7/circuitjs1 #110: Feature: Reset while paused

Felthry, 2017-05-21:
Occasionally, it would be useful to be able to reset the simulation, but not immediately start it. This is especially good for if you want to save something to be used as an example, as you can be sure it will start at zero time with no initial conditions at all. Current behavior is that clicking the reset button sets the simulation to run mode, even if it was paused before resetting.

pfalstad, 2017-05-22:
yeah I've wanted this too.  I think the original reason the reset button starts the simulation is that I was worried first-time users wouldn't know how to start the simulation if it got stopped.  But with the old checkbox replaced with a giant red button, I think this isn unnecessary.

Felthry, 2017-05-22:
It is a nice convenience feature, even if it's inconvenient to not be able to turn it off. Perhaps a separate "reset and run" button? Or just hold the state from when the reset button is pressed, perhaps.

pfalstad, 2017-05-22:
Well, it's not that hard to press reset and then press run.  It's a bit weird to have "Reset" start the simulator if it's stopped.

pfalstad, 2017-05-23:
Actually I could make it so pressing Reset twice starts the simulator.

sharpie7, 2017-05-28:
Fixed.

closed 2017-05-28 by sharpie7