sharpie7/circuitjs1 #113: Bug: Swap terminals has no effect on tunnel diodes, but only sometimes

Felthry, 2017-05-22:
While writing #112 I found that clicking swap terminals on the tunnel diode in this:
$ 13 0.0000049999999999999996 10.20027730826997 50 0.005 99
175 288 192 400 192 0
w 400 144 400 192 0
w 288 144 288 192 0
r 288 144 400 144 0 10
o 0 64 0 4099 0.0048828125 0.000390625 0 2 0 3
o 3 64 0 4099 0.0048828125 0.000390625 1 2 3 3

simulation had no effect whatsoever. However, loading up the tunnel diode I/V curve and attempting to swap terminals works fine. I am not sure what the difference is between the two situations, but both are repeatable on my machine at least.

Felthry, 2017-05-22:
Now I have discovered that the problem seems to be only if your mouse cursor leaves the dropdown menu after right-clicking, then you select the swap terminals button. This is slightly less weird than being dependent on the circuit layout, but still a bug.

pfalstad, 2017-05-22:
yes, I see the bug, it's supposed to flip the menuElm, not the mouseElm..

sharpie7, 2017-05-28:
Fixed

closed 2017-05-28 by sharpie7