sharpie7/circuitjs1 #120: Bug: Resetting default LRC circuit with switch open makes it oscillate

sharpie7, 2017-05-28:
Not clear to me why, or whether this is new behaviour, but:

With the default LRC circuit open, and the switch open, simulation running. Hit the reset button. Circuit seems to be reset in a state where there is stored energy in the LRC loop as it starts oscillating.

pfalstad, 2017-05-28:
This is old behavior; it's even in the java version.  When hitting reset, we don't fully reset.  Instead we put a small charge (1mV) on the capacitors because some oscillator circuits (like the op-amp relaxation oscillator, for example) won't start if everything is reset to 0.

pfalstad, 2017-05-28:
The idea is, many circuits rely on some noise in order to function..  They don't work if everything is totally reset.

altaic, 2017-05-28:
Maybe it would be a good idea to make the noise source an explicit circuit element. Noise sources can be useful for all sorts of stuff.

sharpie7, 2017-05-29:
Fair enough. I might add this to the documentation if I remember!

closed 2017-05-29 by sharpie7

pfalstad, 2017-05-29:
We have noise sources already, but I wouldn't want to introduce them into simple demonstration circuits.  It would be confusing.

It would be better to explicitly switch on this behavior where it is needed, rather than by default everywhere.  It might also be better to add noisy behavior to the transistors and op amps, which is where it shows up in real circuits (I think), rather than putting a charge on capacitors.