sharpie7/circuitjs1 #121: Feature: Offline use

sharpie7, 2017-05-28:
Though not something that I personally want, it has been asked a few times whether it is possible to have the application delivered offline. I have discussed the reasons why offline use isn't supported here under "offline use": http://lushprojects.com/circuitjs/

I note that Electron (https://electron.atom.io/) is starting to become popular. Using Electron may make it possible to have both web and desktop verions using essentially the same code. One thing that isn't clear is whether all the code generated by GWT can be compatible with Electron. Could be a project for someone (not me!) to investigate and test.

Immortalin, 2017-12-03:
@sharpie7 I created a fork specifically for desktop/educational usage. See [Simulacra](https://github.com/Immortalin/Simulacra), you can link to it in the README.

Immortalin, 2017-12-04:
@sharpie7 [first release](https://github.com/Immortalin/Simulacra/releases/tag/v0.0.1), you might want to take a look at [this line](https://github.com/Immortalin/Simulacra/commit/015b2fef78b6984e97454c55eace474807ee3221#diff-8a74dc9011f97cb424f82263f171b3c4R88) if you want to add electron support due to a bug in current electron version.

sharpie7, 2018-01-14:
I have added the configuration files for an Electron app to the repository. Users can now build stand-alone apps using Electron from the source. See the readme.md. There are some limitations currently, but it generally works. At the moment I consider this an experimental feature. Feedback is welcome.

Thanks Immortalin for finding that bug.

pfalstad, 2019-09-21:
Ok I finally took a look at this, thanks Iain!  There is now a link to the mac standalone app on my circuit page.  Windows forthcoming.