sharpie7/circuitjs1 #123: Feature: Possibility of moving to native Javascript?

colinhowell, 2017-05-31:
(Obviously this is more radical than a mere "feature".)

Has there been any consideration of the possibility of converting or replacing this (very nice and helpful) application with a version written in native Javascript, rather than written in Java and relying on Google Web Toolkit? GWT appears designed for larger-scale web-based applications and seems to be overkill for this particular application's needs, and also its long-term future seems to be in some doubt. It would seem to me that a native Javascript implementation, with the help of appropriate libraries, could perform better in the browser and be easier to debug and profile in the long term.

Of course such a conversion would be a lot of work, and I quite understand if either @sharpie7 or @pfalstad are unwilling or unable to devote the time to do it. Or maybe one of you just prefers the Java language. I just wanted to inquire whether a shift might be considered.

pfalstad, 2017-06-01:
Would the performance really be better?  I've heard good things about GWT performance.  I've done some performance tuning on it and the generated code I've looked at seemed pretty good.  I have several ideas for improving the simulator's performance that would be much less work than this.

Debugging and profiling can be a pain, sure.  GWT might be going away in the future, but I'd rather not invest time in translating to javascript until it actually does go away.

colinhowell, 2017-06-01:
To be honest, I don't know if the performance would turn out to be better or not; probably depends on the implementation. And it certainly makes more sense to go after the low-hanging fruit first. I'm sure many, if not most, performance issues here aren't language-dependent.

I just wanted to raise the idea and see if it might be considered in the long term. Sounds like the answer is, "perhaps, depending on how things evolve, but not any time soon". Which is perfectly fine.

colinhowell, 2017-06-04:
@pfalstad, @sharpie7:
(I'm ignoring my original idea, which now feels like it was premature.)

pfalstad, you earlier said, "I have several ideas for improving the simulator's performance that would be much less work than this." Now I'm curious what those are. Could you elaborate?

pfalstad, 2017-06-08:
I can only think of one now and it is basically doing a better job of simplifying large matrices so the matrix solver has to do less work.  The matrix solver is the slowest part of the simulator and is o(n^3) so anything we can do to make the matrix smaller saves a lot of time.  Since the matrix tends to be sparse there are a lot of tricks we can do.  Or maybe we should just be using a sparse matrix solver.  We're already doing some simplification but it could be improved when dealing with nonlinear elements.

closed 2017-08-14 by sharpie7