sharpie7/circuitjs1 #124: Feature: less-resource intensive graphics updating

colinhowell, 2017-06-01:
@pfalstad, @sharpie7:

(This one's a bit of a doozie too, but not as extreme as "maybe rewrite the whole thing in native Javascript?". :D )

For a while I've suspected that this program was a bit of a system hog, at least as far as power utilization was concerned. I'd often hear my laptop fan spin up with it open, even when the circuit itself wasn't running and both Show Current and Show Voltage were turned off.

After studying CirSim.updateCircuit() in CirSim.java a bit, uncommenting the timing lines in there, and hacking the timing code to work the way I wanted, I now understand the problem. updateCircuit() completely redraws everything from scratch every frame: circuit, scopes, the works. Stopping the circuit stops the simulation, but it has no effect on this behavior.

I understand this simplifies the animation of the UI, circuit, and scopes, but the result can be horribly resource-intensive. Right now I have a digital circuit open: a clock, two sets of ganged 4-bit counters, some flip-flops, and some gate networks, 25 elements in all, plus a whole bunch of labeled signals. (If you're curious, it's a representation of the horizontal and vertical counters and sync-generators from Atari's arcade version of Pong.) I have 10 scope traces of various signal voltages stacked on the bottom half of the screen (thanks for allowing the scope area to be resized!), one of which (composite sync) is too dense to be legible at the current time scale and thus looks like a green bar. The circuit is stopped, and both Show Current and Show Voltage are off, but the program is spending *all* its time drawing, my GPU is about 50% busy, and the fan is running pretty hard. (Removing the too-dense signal from the scopes makes things significantly better, reducing the total drawing time by about a third, but even then the GPU is pretty busy.)

I know it would take a lot of effort to optimize the drawing code to only redraw what actually needs redrawing, but it seems like doing so could pay some dividends. It would certainly increase the margin for speeding up simulation before display updates became choppy.

pfalstad, 2017-06-01:
Hi,

I haven't paid much attention to optimizing the case where "show current" and "show voltage" are turned off.  In almost all examples, they're both on. If we're showing voltage, everything has to be redrawn whenever its voltage changes, so you might as well redraw everything.

Clearly we are doing a lot of unnecessary work if both are turned off.  In that case only a few elements need to be redrawn, I suppose.

BTW  I was going to do a simulation of the entire pong circuit at one point, but didn't get very far.

colinhowell, 2017-06-01:
Well, also consider the case where the circuit is stopped and the user isn't modifying anything, regardless of whether Show Current and Show Voltage are on or not. (Perhaps the user isn't currently using the program, but just has it set aside in a browser tab for later. Or they may be simply pausing between edits.) In that case, **all** the drawing is redundant.

In fact, the reason I starting doing things like turning off Show Current and Show Voltage was because I assumed that it would greatly reduce the drawing workload. I didn't realize that it had little impact.

pfalstad, 2017-06-01:
Yeah we should definitely not redraw when the simulation is stopped..  The java version did this, but it got lost in translation to javascript.

colinhowell, 2017-06-01:
So I've been thinking about this problem, and I have some (not fully developed) ideas.

While a single Canvas does not support layering, it does support transparency. You can have multiple layered transparent Canvases with a specified depth ordering, which lets you render in upper-layer Canvases without disturbing lower ones. The browser's rendering engine handles the final compositing step for you.

UI stuff, like the crosshair cursor, scope cursors, and objects being highlighted or moved, would go in the topmost transparent Canvases. When the sim isn't running, only those would need to be updated, and only when mouse events signal a need to do so.

The current and voltage simulation effects could also go in their own transparent Canvases, below the UI Canvases but above the base circuit Canvas. Again, these would only need to be updated if the sim was running and if their respective effects were enabled.

The base circuit Canvas would only need to be redrawn if the circuit was actually changed.

I'm not sure how many Canveses would be reasonable, or whether that would even pose a problem.

Right now you're already using two Canvases, the main one _cv_ and a "back" Canvas _backcv_ where most of the drawing is done, the final result then being drawImage()'d to the main one at the end of each updateCircuit() call. I presume this was to implement a "back buffer" for smoother graphics updates. The "back" Canvas is not itself visible; from what I can see, it isn't bound to any page element. The main Canvas _cv_ is added to the page as the center widget of a DockLayoutPanel. But because the DockLayoutPanel only allows **one** such widget, you can't add multiple overlapping Canvases to that panel. I'm not sure how to work around this, though I imagine there must be a way, either by making a widget which is a container of multiple layered Canvases, if that is possible, or by having the other overlay Canvases being managed outside the DockLayoutPanel while still overlying its center widget. (Obviously I'm not an HTML expert, or I'd already know how to do this.)

sharpie7, 2017-06-01:
I am not convinced this is worth the programming effort. The redrawing makes the graphics part of the program very simple and predictable which is a big plus. The drawing has very little impact on the time taken in the frame loop in my experience. I don't really care if the GPUs have to work. They might as well earn their living. 

Regards 

Iain

Iain

Sent from a mobile device.

pfalstad, 2017-06-01:
Yeah, I didn't think drawing the circuit was that time consuming either.

Step 1 would be to get redrawing to stop when the simulator is stopped.  That's important.

Step 2 would be to reduce the frame rate when current animation is off.  (Right now it's 62fps)  Very easy to do.

If you still have problems after that, we'll see.

colinhowell, 2017-06-01:
OK, I'm just passing ideas along to see if they're worthwhile. If you like them, fine; if not, no problem.

I'm quite happy that this project exists and pleased that it's been rescued from the downfall of in-browser Java. A dynamic interactive circuit simulator is a wonderful tool for better understanding electricity and electronics; one that adds the convenience of running in the browser is even better. The fact that it's open-source just makes it even better, if only for tinkering and experimenting.

closed 2017-08-14 by sharpie7

pfalstad, 2022-03-08:
I stumbled on this issue while looking for something else.  I went back and finished that pong circuit simulation: https://www.falstad.com/pong/