sharpie7/circuitjs1 #128: Help, I don't understand the SR latch IC inputs and outputs in Falstad

jimfoit, 2017-06-16:
First, I'd like to thank Mr. Falstad for an awesome free app.  I think what you've done here will make the world a better place. If there's a heaven you'll make it.

I've searched in Google and YouTube, but can't find any examples.  What are I0, I1, I2, I3, O, O, O, O, and Ld?

I would think there should be a set, a reset, a Q and maybe a not Q.  I've tried connecting wires to the pins w/o any change.  I looked up spec sheets on SR flip flops -- no help.

I'm guessing I0-I3 represent 4 inputs.  But 4 inputs argues for 2 latches in my mind.  But, why then would there be 4 outputs? Obviously, I'm completely missing the ballpark :( 

I'm guessing Ld might be latch output, but I'm too frustrated to randomly try anymore.

pfalstad, 2017-06-16:
Ld is load.  When Ld makes a positive transition, the output pins (O) are loaded from the input pins (I0, etc).  It's not like an sr flip flop..  there's no set/reset.

pfalstad, 2017-06-16:
Thanks for the kind words, glad you like the simulator.

jimfoit, 2017-06-17:
Thank you for the explanation.  I only have about 1/2 a BS in EE that I picked up later in life, so I make lots of mistakes.  But, I'll get there.  It just takes more time.   My designs are to make the poor self-sufficient.

sharpie7, 2017-06-19:
This discussion appears to clarify the issue. BTW, the component is labelled "latch" and not "SR latch" as in the title of this report. The behaviour matches what would be generally understood by the term "latch".

closed 2017-06-19 by sharpie7