sharpie7/circuitjs1 #129: Transformers are bugged?

emilfihlman, 2017-06-16:
Pictures demonstrate situations that are possible to achieve but which do not reflect reality, unless my understanding of electronics is completely bonkers.

[DC current offset with return to zero](https://emil.fi/jako/kuvat/2017-06-16_23-32-18_rkc0p4hY.png)

[DC current offsets with nrz](https://emil.fi/jako/kuvat/2017-06-16_23-33-07_RRY9VyDd.png)

[DC current with no apparent AC component(s)](https://emil.fi/jako/kuvat/2017-06-16_23-35-29_UvYYsBpy.png)

pfalstad, 2017-06-16:
Yes, see here: http://www.falstad.com/circuit/e-transformerdc.html

For a more realistic transformer, use a smaller inductance, a smaller coupling coefficient, and/or more resistance.

emilfihlman, 2017-06-16:
Is there a patch planned?

pfalstad, 2017-06-16:
What do you want patched?  The simulator uses ideal components in general.  If you want less-ideal components, you have what you need to create them.

emilfihlman, 2017-06-16:
>Transformers are bugged?
>Yes

Ideal transformers shouldn't act as pictures demonstrate. In general, not zeroing the currents on reset causes loads of issues with the simulator.

pfalstad, 2017-06-16:
how should it work?  perhaps you can try it out in spice or some other simulator and gather some data about what should be happening instead.

pfalstad, 2019-03-06:
Hard to say, since the images are missing, but I think this was fixed by #162 .  Either that or it is not a bug.

sharpie7, 2019-03-06:
Believe fixed.

closed 2019-03-06 by sharpie7

emilfihlman, 2019-03-06:
Images restored